OpenCPN/OpenCPN #3879: How to work out if really on line?

antipole2, 2024-05-02:
How can a plugin test if OCPN really is on-line?
The API `OCPN_isOnline` looks like what is needed but I find it returns true if connected to  anything, including an on-board network that goes nowhere.
So my plugin thinks it is on-line and tries to download a file from the internet and then a dialogue is displayed allowing the user to abort.
But what is needed is to be able to work out if we are connected to the internet and not just a local network.

How best to do this?

nohal, 2024-05-02:
By trying to download something, there is no other way, what are you allowed to do depends on many things, for example "connection optimizers" like RedPort can filter what resources are available over HTTP very strictly.

leamas, 2024-05-02:
Not an issue but a question. Converting to discussion.