OpenCPN/OpenCPN #3882: NMEA Debug Window Color Coding

torrmundi, 2024-05-05:
**Is your feature request related to a problem? Please describe.**
The color coding of messages in the NMEA Debug Window has two colors that are virtually indistiguishable (black, for Message accepted, and burgundy, for Input message filtered and dropped).

**Describe the solution you'd like**
I would like to change the burgundy to another color that is readily distinguished from all the other colors used. Since currently no green is used, maybe RGB 0,255,0  or 0,128,0 would be good.

**Describe alternatives you've considered**
I used a magnifying glass and increased my screen brightness to maximum.

**Additional context**
[image]

leamas, 2024-05-30:
@torrmundi 

Since the green color commonly is used for "good",  perhaps it would be better to change "Message accepted" to  green ( 0,128,0) and if required also modify "Input Message filtered and dropped" to something else like black?

If I make a patch, are you willing to test it?

torrmundi, 2024-05-30:
Sure., I’ll test it.  My primary concern is being able to tell the colors
apart easily.  Bright sunlight is the worst condition, but at night with
dimmed display can also be tough.

Do we have any other color codes in OpenCPN?  I can’t think of any we
ascribe meaning to like the message status.

John

leamas, 2024-05-30:
I'll see when I find some time. Please stay tuned.

> Do we have any other color codes in OpenCPN? 

None I'm aware of.

sebastien-rosset, 2024-11-29:
In the NMEA message display, when messages appear in red font, the legend indicates `Information Message or Message with errors`. This description is problematic because:

1. It's ambiguous whether the red text indicates an information message OR an error message
2. Users cannot determine the severity or nature of the message from this description
3. The "or" phrasing reduces clarity and could cause confusion in troubleshooting

Looking at the code, I don't see a use case where a message in red means "Information Message".

sebastien-rosset, 2024-11-29:
> Sure., I’ll test it. My primary concern is being able to tell the colors apart easily. Bright sunlight is the worst condition, but at night with dimmed display can also be tough. Do we have any other color codes in OpenCPN? I can’t think of any we ascribe meaning to like the message status.

How about using UTF-8 characters or icon to help represent the type of message, in addition to the color? Basic example below:

[image]

torrmundi, 2024-11-29:
Brilliant!  I think that's a really good idea.

John

sebastien-rosset, 2024-11-29:
I also discovered a non-obvious behavior related to the `LegacyInputCOMPortFilterBehaviour` parameter.

Current Behavior:
The NMEA Debug Window uses two different colors to indicate filtered messages:
- CORAL: Input message filtered, output message filtered and dropped
- MAROON: Input Message filtered and dropped

However, the actual color used for input messages depends on the `LegacyInputCOMPortFilterBehaviour` setting in opencpn.conf:
- When set to 1: CORAL is used for filtered input messages, and these messages still enter the internal multiplexer (available for output connections unless filtered there)
- When set to 0 (default): MAROON is used for filtered input messages, and these messages are completely dropped before reaching the internal multiplexer

Issues:
1. The color distinction is subtle and hard to see in various lighting conditions
2. The behavior difference based on the config setting is not obvious from the UI
3. AFAIK, This setting is only accessible by manually editing opencpn.conf, making it an advanced user feature

sebastien-rosset, 2024-11-29:
In my local workspace, I got this:

[image]

sebastien-rosset, 2024-11-30:
Maybe I'm missing something, but I don't see any code to check the CRC checksum of NMEA0183 message.

rgleason, 2024-12-01:
I like this clarification and hope it is accepted, however do the symbols work in all OS, win, lunux, macos, android?

sebastien-rosset, 2024-12-01:
> I like this clarification and hope it is accepted, however do the symbols work in all OS, win, lunux, macos, android?

Good point. I see the TTY window uses wxFONTFAMILY_DEFAULT, which typically maps to system fonts on each platform:
Windows:
Usually maps to "MS Shell Dlg 2" which is typically Arial or Tahoma
On modern Windows versions, it often resolves to Segoe UI

macOS:
Maps to ".AppleSystemUIFont" which is typically SF Pro on modern systems
On older systems, it would be Lucida Grande

Linux:
Depends on the desktop environment:
GNOME: Usually DejaVu Sans or Ubuntu
KDE: Usually Noto Sans
Other environments might use Liberation Sans or similar

Android (wxQt):
Typically maps to Roboto on most Android devices
Some manufacturers might use their own system fonts

However:
1. The actual font can vary based on system settings
1. Users might have changed their default system font
1. Different versions of each OS might use different defaults
1. The font that actually renders might be a fallback if the default doesn't support certain characters

So I could use something like below and have a fallback if the character is not supported.
```c++
bool IsUnicodeSupported(const wxString& character) {
    wxFont testFont = GetFont();
    return testFont.HasChar(character[0]);
}
```

rgleason, 2024-12-01:
This is beyond my understanding, perhaps Pavel, Alec or Dave will advise.

Hakansv, 2024-12-01:
@sebastien-rosset 
Once you think you're ready for a possible merge I could build and test this PR on some Windows machines.
Tell here when it is appropriate for me to test if you want.