OpenCPN/OpenCPN #3895: Cannot enable OpenGL (option greyed out) on Linux/nVidia system

cleinias, 2024-05-06:
**Describe the bug**
Cannot enable OpenGL (option greyed out) on Linux/nVidia system

I am running opencpn on an archlinux system but I don't get the option to enable OpenGL. The option is always greyed out.

I tried changing the relevant OpenGL options in opencpn.conf, but they are overwritten when opencpn starts up. Opencpn clearly thinks my system is not opengl capable and turns that option of.  Is there a way to check what opencpn's startup script is actually checking? The output to console does not mention any failed test, yet the opencpn.conf file is silently rewritten. The log file  is equally uninformative. 

**To Reproduce**
I am not sure if the bug can be reproduced on different systems, but on the laptop described above, a fresh install exhibits the problem
 
**Expected behavior**
I expect the enable OpenGL option in the settings to be available

**Screenshots**

**Desktop (please complete the following information if applicable):**
 - OS: Linux, Archlinux distro
 - Version: kernel 6.8.9.arch1-1 (i.e. vanilla 6.8.9 kernel)
 - GPU nVidia Quadro1000M

bdbcat, 2024-05-07:
The OpenGL test method has been improved for the upcoming Beta release.
If practical, please try to build from source, and verify the behavior on your system with the new code base.

leamas, 2024-05-07:
There is also the Flatpak packaging which as far as I know  works out of the box on Arch.

cleinias, 2024-05-07:
I just built it from source and there is no difference in the observed behavior. Trying flatpak next

cleinias, 2024-05-07:
Here is the output:

Error: EGL version is 1.4. EGL version 1.5 or greater is required.
{
   "GL_ERROR" : 0,
   "GL_USABLE" : false,
   "GL_ERROR_STRING" : "Missing GL version"
}

Apparently my nVidia driver (390xx) does not provide the correct GL version?

cleinias, 2024-05-07:
Also, the flatpak version is actually worse: even though I can enable OpenGL, no charts are displayed (I just get a solid screen). If I disable OpenGL, the charts reappear

nohal, 2024-05-07:
Your system does not seem to support OpenGL at all. The 1000M should support OpenGL 4.6, so it probably is a driver issue.

cleinias, 2024-05-08:
The GL utilities I can run actually indicate a working openGL environment, though. For instance, here is the output froom glxinfo:

```
stefano@hippias OpenCPN $ glxinfo -B

name of display: :0
display: :0  screen: 0
direct rendering: Yes
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 2048 MB
    Total available memory: 2048 MB
    Currently available dedicated video memory: 1370 MB
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: Quadro 1000M/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 390.157
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.6.0 NVIDIA 390.157
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)

OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 390.157
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```
Is there a chance that opencpn's testing utilities report a false negative?

leamas, 2024-05-08:
However, this seems to be about your system. I'm also on the nvidia driver, although a somewhat more recent hw.  `opencpn-glutil opengl-info` for me: 
```
$ ./opencpn-glutil  opengl-info
{
   "GL_EXT_framebuffer_object" : 1,
   "GL_RENDERER" : "NVIDIA GeForce GT 710/PCIe/SSE2",
   "GL_ARB_texture_non_power_of_two" : 1,
   "GL_SHADING_LANGUAGE_VERSION" : "4.60 NVIDIA",
   "GL_ARB_texture_rectangle" : 1,
   "GL_EXTENSIONS" : [
      "GL_AMD_multi_draw_indirect",
      "GL_AMD_seamless_cubemap_per_texture",
      "GL_ARB_arrays_of_arrays",

      [...]

   ],
   "GL_ERROR" : 0,
   "GL_USABLE" : true,
   "GL_VENDOR" : "NVIDIA Corporation",
   "GL_ERROR_STRING" : "No error",
   "GL_VERSION" : "4.6.0 NVIDIA 470.239.06",
   "GL_OES_texture_npot" : 0
}
```

nohal, 2024-05-08:
As you can see in https://github.com/OpenCPN/OpenCPN/blob/master/glutil/console.cpp#L175 there is no magic in that utility that could go wrong, all it does is it creates an OpenGL canvas and queries the capabilities.

As nobody seems to be able to reproduce your issue, I'm afraid it is you who has to do the digging. From what a quick search returned, https://community.khronos.org/t/opengl-version-is-forced-to-1-4-but-card-can-support-4-6-0/104868/12 looks somewhat similar to what you see.

cleinias, 2024-05-08:
Thanks for your reply. I looked at the link you posted and the issue seems indeed somewhat similar. However, the proposed solution (adding the user to the video group) made no difference, as opencpn still reports no openGL capabilities. I'm trying to figure out if there is a solution on the archlinux forums

nohal, 2024-05-08:
Your architecture is not `arm`, so that part of CMakeLists is completely irrelevant.

cleinias, 2024-05-08:
Then I'm out of luck. My card driver can only provide EGL 1.4 and opencpn requires 15.

nohal, 2024-05-08:
There is probably not much we can do about it, [the actual problem](https://github.com/wxWidgets/wxWidgets/blob/master/src/unix/glegl.cpp#L679) you see in the error message you get is that wxWidgets simply does not initialize the GL canvas without EGL 1.5.
You may try X11 instead of Wayland.

leamas, 2024-05-10:
Anyway,  this does not be an issue about OpenCPN  -- requiring EGL 1.5 is certainly no error. Besides that, we don't  really support native Arch.

@cleinias Are you ok with closing this bug so we can get it off the OpenCPN table?

cleinias, 2024-05-10:
Solution just found with the help of the Arch people: forcing libvgvnd to use the software renderer with by setting a couple of environmental variables, i.e. launching opencpn as follows:

`__GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json opencpn `

This approach seems to work fine on my system. Bug closed

closed 2024-05-10 by cleinias