OpenCPN/OpenCPN #3896: Let output connection "Sentence filter" dialog be default cleared.

Hakansv, 2024-05-07:
When applying a "Transmit Sentence" filter to an output connection the default selection dialog has all sentences selected. Pict 1.
I and most others have to start by "Clear all" since normally only one or more conditions are needed.

Also emphasis the description text to make it more clear the filter function is to restrict transferred sentences to what's selected. Without any applied filter everything available will be sent.

See picture 2.

PR to come if no objections.

Pict 1 Present status:
[image]

Pict 2 Proposed change.

[image]

Hakansv, 2024-05-07:
> probably want all messages sent.

So then don't use a filter and all will be sent. 
I've seen some selected what's listed in the selection dialog as a filter to send all but then you actually have a filter, and send only those listed. So if there are sentences not listed they will not be transferred.
The only reason the use a filter is to get some kind of restriction.

TwoCanPlugIn, 2024-05-07:
What comes to mind is "lipstick on a pig"

I will never understand why we have an option for either an inclusion list or an exclusion list.

No do I understand how an empty "Transmit Sentences" filter means transmit all.

nohal, 2024-05-07:
How do you want to express "Transmit only sentence X" without the inclusion filter?
How do you want to express "Transmit all sentences but X" without the exclusion filter?
With what do you want to replace the "No filtering configured = no filter applied" logic to make it more understandable?

leamas, 2024-05-07:
With questions phrased like that. the current design is the answer.

However, the same result could be achieved with a single list of blocked messages + some new buttons.

  - "Transmit all but X" then is pushing a "Deselect all" button + marking messages messages X to block. 
  - "Transmit only X" translates to pushing a "Select all" button + deselecting messages X to send.

This would probably mean a simpler mental model for user to understand, but the cost is the price of change. This includes upgrade problems.

nohal, 2024-05-07:
This falls apart as soon as new stuff appears in the system. Every time new data appears you will need to revisit the filters.

leamas, 2024-05-07:
Not necessarily, it depends on how data is stored. The two buttons "Select all" / "Deselect All" could be saved, last push wins. When the list of messages is expanded they could be considered.

User model  ≠ database representation

nohal, 2024-05-07:
And how do we express this logic in the GUI in a way that is understandable? I really don't think this has any chance to reduce the, IMHO non-existent, confusion.

leamas, 2024-05-07:
>  I really don't think this has any chance to reduce the, IMHO non-existent, confusion.

This is the real question, and I'm not convinced myself. Need to sleep on it. However, I do think @TwoCanPlugIn has a point.

TwoCanPlugIn, 2024-05-07:
> How do you want to express "Transmit only sentence X" without the inclusion filter?

An inclusion list that only contains X (as it is today)

> How do you want to express "Transmit all sentences but X" without the exclusion filter?

An inclusion list that contains everything except X

By definition, if a sentence is not in the include list, it is excluded.

Or vice-a-versa, using an exclusion filter.
> Transmit only sentence X. 

The exclusion list contains everything except X

> Transmit all sentences but X

An exclusion list that only contains X

> Every time new data appears you will need to revisit the filters.

I believe the filters are populated with a predefined array from the NMEA 0183 library. If there are any new sentences, you will have to update the NMEA libraries. In any case the last NMEA 0183 standard was released IIRC in 2018. I'm not aware of any new sentences.

I am not aware of any commercial software that uses the filter paradigm that OpenCPN uses.  And they have invested heavily in UI design, usability & testing.  For a case in point refer to this recent [thread](https://www.cruisersforum.com/forums/f134/drop-sentences-285926.html)

Hakansv, 2024-05-07:
I really don't understand this discussion. The filter function and logic has been working for many many years without problem.
...
We have a connection to output available messages.
If we don't want it to output all available messages we may use a filter.
To use a filter we have to select conditions. We have two convenient way to set up a possible filter. Either send only those selected OR send all but those selected.
Should we change the text "Output Filter" to "If you want to restrict what will be sent please select filter conditions below" ???

nohal, 2024-05-08:
The "recent thread" is about a mistake of the FR translator who for some reason I really refuse to even try to understand after years of absolutely no problem changed the label to a total nonsense.
 
The "multiplexer" functionality in OpenCPN does not, and does not have to, depend on "knowing the sentences" or the NMEA0183 library. I admit, and would immediately support, that the list of "available sentences" can, and probably even should, be dynamically updated as the data passes the system. That is an idea I haven't yet seen in any of the complaints, but sure is a number one thing on my personal list of improvements that may be viable in this area.
As far as the NMEA0183 standard goes, it is completely irrelevant - a) people abuse it in many ways, b) it allows vendor proprietary sentences we can't ever know about and c) if you generate a list, even just with "known" sentences and talker IDs and their combinations, it will be hundreds of items long. Make it obligatory to use that to implement both the include and exclude scenario in the same way and I'm immediately gone as you are really not serious about trying to make stuff easier for the user.

If we cripple the functionality, we can be like "the commercial software", in Navionics this functionality is completely missing and look how popular it is...

rgleason, 2024-05-08:
> Does the filter list always show the sentences that are selected?

- Yes, it does show the sentences that are selected. .. and the interface appears to be using the same filter storage for accept and apply and transmit and drop.
- When no sentences are selected, there are no sentences shown on the Filter line, and my assumption is that there is "no filter applied"
- Why not inform the user of this directly with "_No filter applied"_ shown in the Filter message area?

rgleason, 2024-05-08:
Which in all cases means "transmit all" or  "drop nothing"  (tranmit all) ?
Or should be radio button be changed to be   Transmit only sentences,  No Filter,  Drop Sentences?
I don't know what is clearer.
Should  "No Filter Applied" actually state something even clearer?

Hakansv, 2024-05-08:
Ok :)

closed 2024-05-08 by Hakansv

rgleason, 2024-05-08:
@bdbcat   The problem is the filter conditions are not shown explicitly to the User.

1. When the Filter menu  has Clear selected, and then the menu is closed. The user does know the condition of the filter, and the filter message line is empty.  Does an empty line mean the filter has been "cleared"   (I think so.)  

2. Can that be stated explicitly somehow?  ...even just below the Message line?

Hakansv, 2024-05-08:
When "filter message line is empty" no filter is applied. All available messages are sent.
As simple as that. What's the problem?

rgleason, 2024-05-08:
Users don't know this. It is not inherently clear that no filter is applied. 
Just because the line is Empty doesn't mean too much to users.
I think it should be stated explicitly. 
"When filter message line is empty, no filter is applied and all available messages are sent"

rgleason, 2024-05-08:
Added Example #3 to both  Input and Output Filtering.
Will add graphic when we  get this page cleaned up.