OpenCPN/OpenCPN #3900: Send to GPS by serial does not transfer unless the connection is disabled.

Hakansv, 2024-05-08:
**Describe the bug**
Send to GPS (serial) does not transfer to the port if the connection in OCPN is enabled.
One need to disable the connection, then it passes through the physical port.

NMEA Debug window and log entries are the same for both occasions. 

**To Reproduce**
Steps to reproduce the behavior:
For test purposes I used a virtual COM6 port receiving on COM5. View COM5 with e.g. Putty
1. Establish a serial output connection with a filter to transfer only "EC". 
2. Enable the connection.
3. Create a route. In canvas menu select "Send to GPS" ("Serial:COM6")
4. In Debug window and OCPN log the route messages are shown as well as "Route transmitted" message.
5. In Putty on COM5 nothing is seen.
6. In connections: disable the COM6 connection. - OK
7. Use canvas menu to Send to GPS.
8. In Debug window and OCPN log the route messages are shown as well as "Route transmitted" message.
9. In Putty on COM5 the route messages are now listed. (See pict)

**Expected behavior**
The "Send to GPS" function should use an enabled serial connection and the applied filter if any.
If no enabled connection (serial in this case) is found the route transfer would not be performed.

**Desktop (please complete the following information if applicable):**
 - OS: Win 10/11
 - Version Git master 5.9

**Additional context**
1. The issue was detected by a user: "When using O alpha 5.9.0 the "Send to GPS" is not working.  If I go back to 5.8.4 it's working.

2. If "Send to GPS" is used from Route and mark manager the described behavior is the same but from there is the progress bar shown. The progress bar is not shown when using "Send to GPS" from Canvas menu. (May be another issue?)

**Screenshots**
[image]

bdbcat, 2024-05-10:
I cannot reproduce in my environment.  But I made some small patch to reduce chances of "collision" with port parameters.  Please re-test and report.

Hakansv, 2024-05-10:
Now the route serial port transfer is performed both when a OCPN connection for the same port is enabled or disabled. The bug is solved. Thanks.
I tested both short and long routes. Below an extract of the long route transfer.

"Send to GPS" function conclusions: 
- "Send to GPS" don't care if a serial(USB) port is listed in OCPN connections. He just "ask" the OS for available serial port.
- Now with the bdbcat fix a serial port also listed and enabled in OCPN connections is useful for "Send to GPS" taking a "back door" to the physical port. That's the "collision" mentioned in the commit.
-  While "Send to GPS" uses the serial port it's also open to transfer what's in OCPM multiplexer and transfer also that. See below last part of the route transfer where not only $ECRTE but also all kind of mux available messages are sent. I suppose this is a side-effect and not a deliberate action and maybe not a problem. It will though steal bandwidth from a 4800 baude rate and prolong the route transfer? 

```
$WIMWV,183.6,R,10.6,N,A*18
$ECRTE,101,99,c,Test route,197,198*76
$WIVWR,180.0,R,8.1,N,4.2,M,15.0,K*65
!AIVDM,1,1,,B,33A@pR7OiB0j1d`Q2p@Ib`2`04sh,0*6F
$ECRTE,101,100,c,Test route,199,200*4B
!AIVDM,1,1,,B,13une470000mrHHPtIoIS0TV0D1q,0*5B
$ECRTE,101,101,c,Test route,201,202*4A
```

closed 2024-05-10 by Hakansv