OpenCPN/OpenCPN #3915: Fonts appear differntly between 5.8 and 5.9 in parallal installation

mguryan, 2024-05-17:
**Describe the bug**
The sizes of the fonts and widgets are wildly differnet when running 5.8 and 5.9 in a parallel installation.
**To Reproduce**
Had 5.8.4 installed and working normally.
Added 5.9.0+9454568
Display of fonts and widget was TINY in 5.9
Reset fonts in Options 5.9
Reropened 5.8. Fonts are HUGE.

**Expected behavior**
Consistent font and widget sizes between versions.

**Screenshots**

5.9:
[image]

5.8:
[image]

**Desktop (please complete the following information if applicable):**
Windows 11

Hakansv, 2024-05-17:
@mguryan 
I may have seen something similar during the progress of 5.9 but now corrected?
The version 5.9.0+9454568 may be old?
To test and verify use the latest alpha from [CS here](https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn_5.9.0-0+1534.939d99d_setup.exe)

mguryan, 2024-05-17:
I uninstalled the previous version of 5.9 and reinstalled the latest alpha. The same behavior is evident. In addition, the icons for waypoints saved in a persitent layer are now oversized in 5.8. Below is a screenshot of 5.9 with User Interface Objects set to 3.
[image]

Below is 5.8 with the same settings:
[image]

bdbcat, 2024-05-19:
This is Windows, yes?
Please report your display size and Windows scale factor employed, if any.

leamas, 2024-06-13:
@mguryan : ping?

bdbcat, 2024-07-05:
Please close or comment further.

closed 2024-07-19 by mguryan