OpenCPN/OpenCPN #3921: Function: "Use WPL position messages" is not working.

Hakansv, 2024-05-20:
**Describe the bug**
If "Use WPL position messages" is enabled and WPL messages are sent neither "APRS position report" nor "Create mark are working,
See more in this [CF report](https://www.cruisersforum.com/forums/f134/opencpn-wpl-aprs-not-working-in-version-5-8-4-a-286142.html#post3901034)

**To Reproduce**
Steps to reproduce the behavior:
1. Enable "Use WPL position messages"
2. Send WPL messages by e.g. VDR (See below)
3. No report or WP created.
4. Also no hit on "else if (str.Mid(3, 3).IsSameAs(_T("WPL"))) {"  in ais_decoder.cpp

**Expected behavior**
Since the function: "Use WPL position messages" is present it would work

**Screenshots**
See CF post linked above.

**Desktop (please complete the following information if applicable):**
 - OS: Tested on Win 10
 - Version 5.9...

**Additional context**
Tested NMEA messages:

```
!AIWPL,5054.40,N,00437.53,E,ON65SX*5B
$GPWPL,5057.68,N,00358.34,E,ON7GR*68
$GPWPL,5112.60,N,00336.76,E,OS5P*34
$GPWPL,5036.95,N,00300.75,E,F5SSM*78
```

TwoCanPlugIn, 2024-05-22:
FYI, Rudimentary testing of NMEA 0183 DSC messages.
Fortunately no regressions found with my limited testing :-)

TwoCanPlugIn, 2024-05-22:
Little, other than both are/were processed somewhere in the bowels of ais decoding and I was worried about yet another regression.

bdbcat, 2024-05-23:
"DSC is setup to be handled by the observer event but WPL was not."
That is the essence of the problem.  but there is more...

bdbcat, 2024-05-26:
Complete, close after testing (with VDR).

Hakansv, 2024-05-26:
Yes it works but we have, for any historical reason, an optional use for this function.
Condition for that included in PR 3933

PR #3933 (2024-05-26): Add condition for the optional "Use WPL position report...."
Closes #3921

Also clean out old code not used anymore.
closed 2024-05-26 by bdbcat