OpenCPN/OpenCPN #3922: Modal "device not found" dialog blocks boot.

leamas, 2024-05-20:
**Is your feature request related to a problem?**
If a device configured in a connection is not found on boot there is  message box like _Device /dev/ttyUSB0 not found (disconnected?)_.  This is a modal message, and until user pushes the OK button positions are not fed to existing devices.

While not a problem on a normal device, this is a problem on a headless one since it means that user needs to connect a PC and clear the dialog box before positions starts flowing.

**Describe the solution you'd like**
The best is probably to make the modal dialog modeless which means that it would be visible as today but positions and other data flows in parallel with the dialog. 

**Describe alternatives you've considered**
 A simpler approach would be to have a timeout, but this might hide a quite important message in a headless context.

leamas, 2024-05-20:
The same is probably true for the dialogs presented  for missing device permissions on Linux. However, this is less likely to happen in a correctly configured environment.

bdbcat, 2024-05-22:
We actually have a method to produce a timed YES/NO message dialog.  This is useful for startup dialogs in a headless environment.
I think it may be unused at the moment, though. Hmmm

leamas, 2024-05-22:
Yes. But using that would mean that in the headless case a user which connects later will not  see this quite important message. 

It would also mean a long delay before positions start flowing after startup

PR #3932 (2024-05-26): udev_rule_mgr: Make "device not found" dialog modeless  (#3922).
Instead of a modal dialog which blocks the startup use a modeless frame. Track the open instances to make it possible to destroy them if they still exists whne exiting opencpn.

Closes: #3922
commit 8844a38e (2024-05-26): udev_rule_mgr: Make "device not found" dialog modeless  (#3922).

Instead of a modal dialog which blocks the startup use a modeless
frame. Track the open instances to make it possible to destroy them
if they still exists whne exiting opencpn.

Closes: #3922
closed 2024-05-26 by leamas