OpenCPN/OpenCPN #3923: NMEA Debug WIndow: Confusing event handling

leamas, 2024-05-20:
**Is your feature request related to a problem?**
When opening the NMEA Debug Window it presents some buttons like _Pause_ and _Copy_. However these buttons are not usable, there is no way to invoke these actions. Likewise, it is not possible to close the window using "normal" methods like the top-right X. 

After closing the options window which launched the debug window, buttons and top-right X works as expected. This is a plain mess confusing all first time users, and also real far from all known UX guidelines.

**Describe the solution you'd like**
All functions in the window should be available form start.

**Additional context**
The root problem is that `class options`  which is used to launche the window is modal and thus "steals" all events. The debug window is thus  not attached to any main loop until the options window is closed.

Non-functional GUI elements are arguably a bug. However, this is debug functionality  which does have lower prio. Hence: "enhancement".

Hakansv, 2024-05-21:
Just for info, not underrating the Linux disadvantage:
On Windows you can handle all NMEA Debug window functions also when Options is open.

leamas, 2024-05-21:
> On Windows you can handle all NMEA Debug window functions also when Options is open.

Which arguably is a wxWidgets bug. `g_options` is run using `g_options->ShowModal()` , and ShowModal is [documented](https://docs.wxwidgets.org/3.2/classwx_dialog.html) as _A modal dialog blocks program flow and user input on other windows until it is dismissed_.  

The reason for the Windows behaviour might be that the debug window is a separate, modeless wxFrame window launched from the modal options dialog. This might be more or less undefined , it is a strange thing thing to do.

bcn2, 2024-05-23:
The same bug applies to macOS as well.
Is there since several years.

bcn2, 2024-06-18:
On macOS the NMEA debug windows is ok with the actual beta.

leamas, 2024-06-18:
The debug window is OK on all more capable machines. However, it breaks things on the RPi.

leamas, 2024-06-18:
The fixes aimed to close this issue are now reverted since the caused #3987.   Which also means that this bug reappears and most likely won't be fixed in 5.10.

For the record: the reverted commits: 0b475d2d, c2c5b0e and 413b2725.

Problems encountered:

  - z-order issues, in particular on MacOS, making the debug window sink under the options dialog. Probably fixable by creating the window as a child of the dialog, reparenting it to the global frame if/when the dialog is closed.
  - Some crash when the main flow continues without waiting for a modal dialog to complete. Needs to be fixed
  -  The solution in the patch series used 100% of a processor to poll for the main thread to complete. This breaks at least  the RPi.
  - Overall, a modeless options dialog supports a modern style "apply when clicked" rather than  the oldOK/Apply/Cancel approach (which brings a really poorly defined "Apply" operation).
  - A simple fix would be to keep the dialog modal and postpone opening of the debug window until the dialog is closed. No good UX, for sure.

leamas, 2024-12-10:
As stated in #3923: there is no point making minor fixes to the nmea debug window. It needs a total rewrite.

Closing

PR #3924 (2024-05-23): Fix NMEA log window event handling (#3923)
  - Make options to sort of a "semi-modal" dialog which allows for the NMEA log window to process events.
  - Hook up the NMEA log window close event  so  the "Show NMEA Debug window" is updated when user closes window.
  - While on it: some general clean up i affected files.

Closes: #3923
closed 2024-12-10 by leamas