OpenCPN/OpenCPN #3925: Build Error in wxVector Implementation

bensasl2, 2024-05-23:
I'm encountering a build error while compiling the OpenCPN project. The error occurs in the wxVector implementation within wx/vector.h, specifically in the push_back method.

[image]

OS: Windows 11 64-bit
Compiler: MSVC 2022
Build Tool: msbuild
wxWidgets Version: [3.2.5]
OpenCPN Version: [5.9.0]

I would greatly appreciate any guidance or suggestions on how to resolve this issue.

Thank you for your help!

rgleason, 2024-05-24:
Did you setup your environment following

https://opencpn-manuals.github.io/main/opencpn-dev/windows.html#_complete_builds_using_visual_studio_2022_community_edition_or_visual_studio_2019_community_edition

leamas, 2024-05-24:
This is not a bug in wxVector. It has been discussed in [zulip](https://leamas.github.io/zulip-archive/stream/332168-Master---5.2E8.2E0--.28was.3A-comms.29/topic/Windows.20build.20broken.html)

transmitterdan, 2024-05-24:
To my knowledge it only fails for the debug build type. It works for non-debug builds.

bensasl2, 2024-05-24:
Thank you for the heads up here and on zulip, happy to work with RelWithDebInfo configuration for now.
What s the bets option to sort the usual missing DLL issue when I run from VS2022 ? 
Thanks !

transmitterdan, 2024-05-25:
I am not aware of any missing dll issue when using winconfig.bat to configure and build O.

Hakansv, 2024-05-25:
@bensasl2 
You either have to use winconfig.bat to set up your system or if you for any reason did it "by hand" you have to copy all runtime dll's from, my example, "C:\Program Files (x86)\OpenCPN\" to "C:\Builds\OCPN\OpenCPN\build\RelWithDebInfo\"

bensasl2, 2024-05-25:
I changed winConfig.bat to exclude Debug build. I was then able to run from VS2022 with no issues .
thanks

bdbcat, 2024-05-28:
Please close this issue if resolved.

transmitterdan, 2024-05-28:
Dave,

No one can build a debug version on VS2022. The non-debug builds will complete and run ok. I think it is a compiler bug but not 100% sure.

The file that fails is shapefile_basemap.cpp. It's been this way for several weeks. Compiler complains about wxVector and the error comes from wxWidgets. There appears to be nothing wrong with OpenCPN source code.

I would like to leave it open until we can get debug builds to complete and run.

Dan

transmitterdan, 2024-05-28:
See PR: https://github.com/OpenCPN/OpenCPN/pull/3943

closed 2024-05-29 by leamas