OpenCPN/OpenCPN #3926: OpenCPN corrupts navobj.xml.changes and then will not start

svMigration, 2024-05-24:
**Describe the bug**
I close OpenCPN by clicking the X in the upper right corner. OCPN does not shut down properly. But this is not clear. A crash report is not generated. When trying to start OpenCPN again, the safe restart message is displayed (this is how I know OCPN did not shut down properly). Regardless of whether I choose Yes or No for safe restart, OCPN displays the chart for about 7 seconds and then shuts down.

If I remove the navobj.xml.changes file from C:\ProgramData\opencpn then OCPN opens just fine after answering NO to the safe restart message.

**To Reproduce**
Steps to reproduce the behavior:
Can only reproduce if OpenCPN shuts itself down improperly
1. Replace navobj.xml.changes file in 
C:\ProgramData\opencpn
with navobj.xml.changes file found in
[navobj.xml.zip](https://github.com/OpenCPN/OpenCPN/files/15425489/navobj.xml.zip)
2. Start OpenCPN

**Expected behavior**
OpenCPN will show screen for 7 seconds and then close.

**Desktop (please complete the following information if applicable):**
 - OS: Windows 11
 - Version 22H2
 - OCPN version Release 5.8.4

**Additional context**
Probably half of the time I close OpenCPN down, I get the safe restart message when I start again. There are no crashreports generated. Usually after OCPN has been running for several hours and tracking is on. 

Log file from today's episode is here:
[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/15425519/opencpn.log)

Hakansv, 2024-05-24:
A short test on O 5.9 from today's Git master. No error found here.
There have been some work on this recently. Maybe it's fixed?
What I did:
I saved you example xml to "C:\ProgramData\opencpn\"
Started O; runs as usual.
Stopped O. The XML-changes file was silently removed.
Started O; runs as usual.

leamas, 2024-06-13:
@svMigration  The beta is out. Could you possibly test it to check that the issue indeed is gone? 

Info on installing the beta [here](
https://www.cruisersforum.com/forums/f134/opencpn-v5-9-1-beta-test-starts-286845.html)

svMigration, 2024-06-17:
@leamas  I installed the beta. Opened. Worked fine. Closed. Copied the corrupted navobj.xml.changes to C:\ProgramData\opencpn. (there was no navobj.xml.changes in the directory so did not replace one). Opened 5.9. No problem. So looks like the issue is fixed. Thank you.
I'll keep running the beta to test but we won't be heading out on our next voyage for at least a week.

closed 2024-06-17 by svMigration