OpenCPN/OpenCPN #3927: Geographic Filter For AIS

valschmidt, 2024-05-24:
**Is your feature request related to a problem? Please describe.**
Starlink makes it possible to subscribe to global AIS feeds (aishub.com, marinevesseltraffic.com, etc.) rather than just the data that one is able to receive with one's own shipboard receiver. This is great because on smaller vessels capable of supporting only small antennas it can be difficult to receive transmissions from distant ships. However, OpenCPN becomes sluggish and more often than not crashes with this global volume of data. 

**Describe the solution you'd like**
Two options added to the GUI for setting up incoming AIS data feeds:
1) Filter and omit AIS messages with object locations outside a user-specified range from some fixed point.
2) Filter and omit AIS messages with object locations outside a user-specified range from my vessel. 

**Describe alternatives you've considered**
Writing my own code in python to do the filtering before sending the data on to OpenCPN. But this is silly and hard to maintain and transport.

**Additional context**
There may be a need to omit some message types from this filter. For example, if one were using OpenCPN for voyage planning, one might want DTONs and other fixed objects to be displayed even if they fall outside the filter. But one would probably still not want all of these the world around. Probably best to implement the basic feature and see what user's think.

Hakansv, 2024-05-24:
My personal comment.
To use AIS data from marinetraffic or similar sources would not be suitable for a navigational system. Data can be old and not updated. Instead real data from a local AIS receiver is the true story. Even a small antenna would cover what's needed around you boat for CPA and other features.
Also a range filter inside OCPN does have to handle each and every incoming data that may disturb you navigation system, as you mentioned.
To see the the whole world's AIS targets I should have used a PC/tablet not used for navigation.

Hakansv, 2024-05-24:
> In our experience a small antenna on a small boat is not enough

I've for many years used a small, 1/4 wave, VHF antenna mounted not so high om my class B AIS. 10 to 20 Nm are well covered. Class B with "hidden" antennas are some times restricted to only a couple of Nm but they would probably neither being seen by a distant shore receiver reporting to MT?
If you've trouble with a "small" antenna try to mount it higher up, or check the cable for noise.