OpenCPN/OpenCPN #3928: Check Chart Panel > Show chart data quality does not work

rgleason, 2024-05-24:
**Describe the bug**
When **Chart Panel > Show Chart Data Quality** is checked the proper icons do not show.

However, when I check **Toolbar Options > Chart > Vector > Chart Object Display**, only then are the proper icons shown.

**To Reproduce**
As stated above.

**Expected behavior**
When **Chart Panel > Show Chart Data Quality** is checked the proper icons should show, and
a message should pop up, stating that 
 **Toolbar Options > Chart > Vector > Chart Object Display** has been checked or turned on.

**Desktop (please complete the following information if applicable):**
 -Windows 11
 - 5.9

Hakansv, 2024-05-24:
Just to compare on Win10 and 5.9 current master.
I can toggle it on and off, see pictures.
Pls explain more about the "message". What's that?
[image]
[image]

rgleason, 2024-05-24:
Hakan, when chart objects is off, checking  show chart data quality does not work!

That is it.

If chart objects must also be turned on, then it should do so automatically!   

Users do not know about these things.

The message is just an idea to let the user know what else has been turned on.

rgleason, 2024-05-27:
This is profoundly screwed up, and writing it up for the manual makes no sense to me at all, in fact I won't even try until this is fixed.  
What is the meaning of "Chart information Objects"?  What is it exactly please?
What is the meaning of "Chart Data Quality"  
I think  that  "**Chart Data Quality**"  =  "**Chart information Objects**" , but I certainly could be wrong about that.
If I am right, then we have two separate Checkbox Settings  in two different menus (Vector Chart Display and Chart Panel) that do not work properly and in unison.
In fact, one of the Checkbox Settings is unnecessary.
Please explain exactly how these are supposed to work.
...very confused.

[image]

[image]

[image]

rgleason, 2024-05-27:
Yes, Hakan  Is the "Chart Information Checkbox" needed any more?
I vote NO.

rgleason, 2024-05-27:
Once this is done, I will make new screenshots and move the "Chart Information Objects" from the Vector Charts Display page, over to Chart Panel  > Vector Charts  where the Checkbox "Show Quality of Data" is located.

I will then paste the entire "Chart Information Obects" writeup and change the title to  "Show Quality of Data"

bdbcat, 2024-05-27:
Not so fast, here.
The tag "Chart Information Objects" refers to object classes whose name begins with "M_"
See this link to discover what the set of such objects are:
[https://desktop.arcgis.com/en/arcmap/latest/extensions/maritime-charting/s-57-object-finder.htm](url)

You will see that M_QUAL is just one of the possibilities.  
So, "Chart Information Objects" is not technically equivalent to "Show Quality of Data".

I can understand the confusion in the current UI.  Needs improvement.  But not scrapping.
The proper solution would be to clear and "grey out" the "Show Quality of Data" checkbox if  "Chart Information Objects" is NOT checked also.

Also note:  Just because you see no "Chart Information Objects" on the ENC cell viewed (except maybe M_QUAL) does not mean they are not available for some other chart.  NOAA is a bit sparse on information objects.  Other publishers are more verbose. 

All of which goes to show that S57 representation of charts is complex.  Reducing complexity so that novice users understand organically, without any further study, is not a good policy.  We must teach. as well.

Hakansv, 2024-05-27:
Many thanks for the clarification. I obviously didn't looked close enough in the code to understand it. 
The link to S-57 (M) objects made it clear.
If M-QUAL toggle function is greyed out also by "Chart Information Objects" it would be consequent to when the level of Display Category does it. 
It may still not be obvious for all users why it's greyed out? And to make the toggle function and the Options parameter texts enough explanatory is not easy.  "Chart Information S57 'M'-Objects" ???

Although this wouldn't help a user on the ocean but it's what we can do I think: @rgleason can you with a well-formulated short note in the manual make this clear? Maybe use Dave's link.

rgleason, 2024-05-27:
What about having Show quality of data turn on Chart Information objects?

Do we know whar is in the set of "Chart information objects?

Does the Display Category choice, change operation of Show Quality or Chart Information Objects?

Furthermore, I thought that Chart Panel was to provide quick and easy access to settings most needed.

Having a setting that doesnt work does not make sense.

Having a popup message, Turn on Chart Information Objects in Vector Chart Display does at least give the user a clue, but that is bizarre in itself, and to complicate things the user might have set Display Category to something that changes all this. 

As I recall, I also had to physically check Quality of data in the Display List as well (I may be wrong about this) but there are so many things involved here, how do you explain it when it makes no sense to the user?

rgleason, 2024-05-27:
Its a blocker because this part of the manual can't be written.

bdbcat, 2024-05-27:
Rick's idea has merit.
If user clicks "Show Data Quality" (SDQ) "on", and "Chart Information Objects" (CIO) happens to be "off", simply turn on CIO, and leave it on.
That is,  when user clicks SDO "off", leave CIO "on".
This should do a sensible thing for a basic user. I would tend to think viewing and analysis of full CIO set as an advanced feature, anyway.

rgleason, 2024-05-27:
Thanks very much. I will make those notes/changes to the manual. Let me know when you'd like this closed.

rgleason, 2024-05-27:
Display Category  |  Show CDQuality  | CIObject |  Object Listing | ENC Chart shows
Base                     |  Greyed out |  on/off no change in chart |  cannot scroll, not available to see
Standard               | Greyed out | on/off  no change in chart |  cannot scroll, not available to see
All                         |  off |  off   |   off   |  Quality of Data off, can scroll | CDQ does not show  
All                         |  on  |  off  |   on  |  Quality of Data on, can scroll  | CDQ does not show
All                         |  off |   on  |  off   |  Quality of Data off, can scroll | CDQ does not show  
All                         |  on  |  on  |   on  |  Quality of Data off, can scroll  | CDQ not show
User                    Similar to ALL

Question: When "BASE" is selected for a Chart Category, in Options > Vector Chart Display
Aren't we supposed to be able to scroll the User Standard Objects List?
I can't do that, the only thing I can do is enlarge the window to see more of the settings.
[image]

Same thing happens with Standard.
For Base and Standard, are the  List of Objects supposed to be greyed out?

I wanted to see what happens when I toggle  CIO.

rgleason, 2024-05-27:
This is definitely an avoidable mess. See
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:chart_panel:chart_panel_options:vector-charts

bdbcat, 2024-05-28:
Rick...
As you have discovered, CIO is only visible in "ALL" or "USER" display category.
Always been that way, and follows IMO recommendations.

bdbcat, 2024-05-28:
Question: When "BASE" is selected for a Chart Category, in Options > Vector Chart Display
Aren't we supposed to be able to scroll the User Standard Objects List?

No, the list is only editable in "User" category, not in Base, Standard, or All

bdbcat, 2024-05-28:
Rick...
Improved UI interaction of CIO and SDQ.
Please test and close if satisfied.

TwoCanPlugIn, 2024-05-28:
Kind of related to the overall intent of this [discussion topic.](https://github.com/OpenCPN/OpenCPN/discussions/3773)

Another example of options that should be greyed out (disabled) depending on the setting of another option.

rgleason, 2024-05-28:
Good morning.
Bdbcat wrote:
> Question: When "BASE" is selected for a Chart Category, in Options > Vector Chart Display
> Aren't we supposed to be able to scroll the User Standard Objects List?
> 
> No, the list is only editable in "User" category, not in Base, Standard, or All

Yes, I understand  Display Categories  BASE, STANDARD & ALL are not editable. (I was thinking about this last night.)
BASE  "User Standard Objects" checkmarks  are greyed out (for unselectable).
STANDARD  "User Standard Objects" checkmarks are blue, (also unselectable)  (we can see the difference between BASE and STANDARD)
ALL "User Standard Objects"  are blue (also unselectable )(we can see the difference between BASE and ALL

Comments: 
1. Why not let user the scroll the menu up and down (to see the objects in the set and be understand the comparisons)?
2. Wouldn't it be more appropriate to use a darker version of grey instead of blue? (using tones of grey as "unselectable")?

Thanks, I will find the most recent on your cloudsmith and test.

rgleason, 2024-05-28:
Dave,  It tests nicely for Display Categories ALL and USER STANDARD . Very simple  check/uncheck of SDQ in Chart Panel turns on and off the M_QUAL in the charts.
In the background TB > Options > Charts >Vector Charts CIO is checked if needed to show M_QUAL.
In the background when SDQ is unchecked, TB > Options > Charts >Vector Charts CIO is left checked which is fine I think.

Should I make my two comments above into RFE?

closed 2024-05-28 by rgleason

bdbcat, 2024-05-29:
RFE is fine, but understand that we have no control over "shades of gray" disabled indication.  Color is determined by wxWidgets/OS.