OpenCPN/OpenCPN #3929: RFE:  Flexible Regex Filtering

rgleason, 2024-05-24:
Flexible Regex Filtering
NOT IMPLEMENTED

Would allow more flexible Regex Filtering.
Background:

Often, there is more than one source sending position information, for example $GPRMC and $IIRMC.
The latter provides rather poor accuracy and needs to be rejected, otherwise the boat will “jump around” on the map.
But other data from this source, like $IIDBT is needed. With the filtering currently implemented, it is quite cumbersome to find out the “bad guys on the bus” and do the job.

With this proposed extension, filtering will be possible by inserting in the Opencpn.ini or Opencpn.config section [Settings/NMEADataSource]. Simple wildcard patterns like

 ;.II[GZ]..,.IIRM.;

at the location where the filter strings are stored.

The matching will only apply if the pattern does not have 2, 3 or 5 characters. \\This can be accomplished by inserting dots as wildcard characters, as in the above example.

Later, the GUI settings dialogue may be adapted to support this and prevent incorrect input.

nohal, 2024-05-25:
With the current filtering it is absolutely trivial to filter out $IIRMC, you simply create a filter `$IIRMC`, given the size of the set of NMEA sentences and talkers is very limited, regular expressions in the filters with all their complexity make pretty much no sense.

Hakansv, 2024-05-25:
There's no need to filter that case.
We have the well functional source prioritization where you can control which of them shall be used for position.
In the described case, GPRMC and $IIRMC, there would have been no "jump" since the first received would have been used.  If that fails the second will be used instead, until the first come back.
The user has all time the possibility to change which shall be first priority.

rgleason, 2024-05-25:
Ok, good points,, so this is a thought from before that found its way into the manual. There is no need for it. I am closing.

closed 2024-05-25 by rgleason