OpenCPN/OpenCPN #3930: Small bug: Check Show FPS  does not work

rgleason, 2024-05-25:
**Describe the bug**
[Options > Display > Check Show FPS](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:set_options:display#show_fps)  does not show in the menu bar at the bottom. 

Windows 11 and MacOS
Opencpn 5.9

Hakansv, 2024-05-25:
Yes I see the same.
What I can see in the code this function is only valid for Linux when O is started from a command prompt, then FPS are printed in the command window. Nothing is printed to O chart canvas.
Pls Linux gurus correct me if I'm wrong.
If I'm correct the fix would be to show this option only when O is started as described above? Or dismantle it?

leamas, 2024-05-25:
> f I'm correct the fix would be to show this option only when O is started as described above? Or dismantle it?

Or fix it so it writes on the canvas.

Hakansv, 2024-05-25:
Who need it and for what reason?

bdbcat, 2024-05-27:
It should be removed.  Measuring FPS on parallel GPUs using GLSL is notoriously inaccurate, and has been generally deprecated in modern apps..
I got this.

closed 2024-05-27 by bdbcat

rgleason, 2024-05-27:
Thanks, I will remove reference to this in the manual or mark "deprecated."

rgleason, 2024-05-27:
I didn't close this.

commit 1d778596 (2024-05-27): Remvoe OpenGL option "Show FPS", Closes #3930