OpenCPN/OpenCPN #3931: UserIcon for marker appears much smaller

ukoehler, 2024-05-25:
An icon used for a marker from UserIcons appears much smaller.

Steps to reproduce the behavior:
Copy /usr/share/opencpn/uidata/markicons/Activity-Diving-Snorkel.svg to ~/.opencpn/UserIcons/
Change the color and rename to Activity-Diving-Snorkel-red.svg
Start OpenCPN
Change Icon of Marker from Activity-Diving-Snorkel to Activity-Diving-Snorkel-red
Icon appears much smaller when hitting ok

Restart OpenCPN and the icon are the same size

**Expected behavior**
No restart required.

Have the icon the same size. I also tried to change the width and height in the SVG from 18mm to 64. Tried a 64x64 PNG, as well. Always the same result

Expect the marker icons to appear the same
**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Linux KDE Neon Almost up-to-date (one week behind latest updates)
- OpenCPN version: 5.8.4 .. ubuntu22.04.1

bdbcat, 2024-05-28:
Temp hold on this issue pending verification on O5.9 Beta.

leamas, 2024-06-13:
@ukoehler The beta is out. Could you possibly test it to check that the issue indeed is gone? 

Info on installing the beta [here](
https://www.cruisersforum.com/forums/f134/opencpn-v5-9-1-beta-test-starts-286845.html)

ukoehler, 2024-06-15:
I am currently very busy, but I certainly will look into it.

The version I had: opencpn-5.8.4+8089+1637c28fb~ubuntu22.04.1.amd64 (manual:lp_ppa_opencpn_opencpn-jammy-main)

Just tried again with the old version and now I see a small red icon and a large black one. So I can reproduce the problem.

ukoehler, 2024-06-15:
No, I still get the same behaviour. That might be due to the SVG. In the selection list for icons it is also displayed in the smaller size. I had similar issues with AvNav and SVG icons.

I had a look at the icons and the version saved by Inkscape after changing just the color looks quite different. I manage to eliminate the problem by changing    width="64" height="64" to width="18mm" height="18mm".

It would be enough to just document that behaviour in my view.

rgleason, 2024-06-16:
Changed manual see https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:faq:user-interface-trouble#usericon_for_marker_appears_much_smaller

ukoehler, 2024-06-17:
Sorry, I tried to replicate my results and  could not. Also, that paragraph sound strange without the other text of this issue. May I therefore suggest the following text:
"UserIcons might appear a different size in the selection list and after first using them. This effect will be fixed by restarting OpenCPN."

rgleason, 2024-06-18:
Have changed it.
 Basic User Manual FAQ. see https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:faq:user-interface-trouble#usericon_for_marker_appears_much_smaller

bdbcat, 2024-07-05:
Close this now?

rgleason, 2024-07-05:
The original poster should test and close this issue if resolved. That is the  accepted  process.

ukoehler, 2024-08-04:
The change in the manual is ok as far as I am concerned. I would still be interested if somebody could point me to the points where the marker icons are used in the code, please? There is some strange behaviour (selection box and usage in the map display) that I would like to investigate. It might well be in underlying code

bdbcat, 2024-08-04:
Uwe....
Search in the code for 
` void ProcessUserIcons(ocpnStyle::Style *style, double displayDPmm);`
This will get you started.
Keep in mind that icons are processed once on initial install, then stored in a cache for later access.
Were I debugging this, I would disable this cache, set some breakpoints, and step through.

closed 2024-08-04 by ukoehler