OpenCPN/OpenCPN #3934: Changing screen scale in Gnome Settings Display corrupts viewport coordinates.

marfree, 2024-05-26:
**Describe the bug**
When altering screen scale, the resolution is maintained, however the display area is cropped or extended. OpenCPN must be shut down but chart bar 

**To Reproduce**
Steps to reproduce the behavior:
1. Using Gnome Go to Settings>Displays, adjust the scale up to improve ui visual accessibility, 
2. Launch OpenCPN, configure a split window of MBTIles charts using Google Earth and Navionics
3. Go back to Settings>Displays and readjust the scale down to improve chart real estate.
4. See truncated viewports, chart bar is non functional, messed up.

## Report details
- **Date generated:**                              2024-05-26 02:50:44

## Hardware Information:
- **Hardware Model:**                              AZW GTR
- **Memory:**                                      64.0 GiB
- **Processor:**                                   AMD Ryzen™ 9 6900HX with Radeon™ Graphics × 16
- **Graphics:**                                    AMD Radeon™ Graphics
- **Disk Capacity:**                               3.0 TB

## Software Information:
- **Firmware Version:**                            GTR6_P5C4V16
- **OS Name:**                                     Fedora Linux 40 (Workstation Edition)
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               46
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.8.10-300.fc40.x86_64

leamas, 2024-05-26:
I  assume that since you are on Fedora you are running Flatpak?

(writing this comment on Fedora 40 as well)

marfree, 2024-05-26:
Yes, it was installed via Software 46.1. The default is a flatpak from flathub.org

[image]

leamas, 2024-05-27:
I don't see any platform independent way to solve this (?). 

In the general case this should supposedly be handled by changing the opencpn  internal resolution.  There has been some work done on this in this cycle, it would be interesting if you could test the Real Soon upcoming  beta version using this technique.

For Flatpak/Linux it **might** be possible iff there is some dbus event generated by Gnome when changing scale which  can be acted upon. I have no idea  if there is such a dbus interface. Have you?

leamas, 2024-05-27:
There seems to be *org.gnome.Mutter.DisplayConfig* which also has a signal for changed settings. This will  probably work on most distros besides Raspbian which does not use mutter  but wayfire.

That said, I'd say doing anything about this is likely to be deferred for now.

leamas, 2024-05-28:
Adding a proof of concept patch which listens to Wayland/Mutter  dbus events for changed monitor setup and notifies an EventVar.  The test listener does nothing but prints a string. Based on this a full solution could be made by

1. Re-scale the canvas in the listener in the same way as when it is created, whatever that would mean.

2. Possibly add support for other platforms listening to platform signals and notifying the EventVar in the same way as Linux/DBus.

[0001-WIP-dbus_client-Add-SendGetMonitorConfig.patch.gz](https://github.com/OpenCPN/OpenCPN/files/15464812/0001-WIP-dbus_client-Add-SendGetMonitorConfig.patch.gz)

leamas, 2024-05-28:
@marfree 

>  Chart bar on restart with a scale of 200% set prior will not allow you to hover and click charts to load.

Is this #3612?

leamas, 2024-06-02:
Re-adding deferred.  This issue is triaged and is indeed deferred

leamas, 2024-07-24:
@marfree: Could you possibly test the current beta w r t this issue and report back? Instructions [here](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_advanced:program:flatpak_qanda#qhow_do_i_install_a_flatpak_beta_release)

Note that the beta is installed in parallel with current installation,

leamas, 2026-09-15:
Sidenote: I cannot reproduce this on Fedora 43/Flatpak/org.opencpn.OpenCPN//stable  using wayland. On the contrary, opencpn works fine even if scale is changed while it's open.

leamas, 2026-09-19:
Since I cannot reproduce it and there is no feedback from the OP I am closing this. There is just not information enough to do anything about this.

So, closing. Please write in closed bug  if there is a need to reopen it.

closed 2026-09-19 by leamas