OpenCPN/OpenCPN #3937: Bug: Important Text Only = No Text

rgleason, 2024-05-27:
**Describe the bug**
When Important text only is checked  --> No text.

**To Reproduce**
See Screenshots

[image]

[image]

Using Noaa ENC

**Expected behavior**
At least there should be some text.

**Desktop (please complete the following information if applicable):**
 - Win 10
 - Version 5.9

bdbcat, 2024-05-27:
The filter works as expected.  If a text element is not specified _by the cartographer_ to be "important", then it is not shown.
NOAA typically applies the "Important" tag the fixed bridge air clearances, and not much else.
See pix at New Bedford
[image]

rgleason, 2024-05-27:
Thanks, closing

closed 2024-05-27 by rgleason