OpenCPN/OpenCPN #3938: opengl is disabled on opencpn v5.9 portable

Mohamed-ElDesoky, 2024-05-27:
opengl mode is disabled on opencpn v5.9 portable although the opencpn-glutil.exe is exist

nohal, 2024-05-27:
Post the logfile.

Mohamed-ElDesoky, 2024-05-27:
[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/15453885/opencpn.log)

nohal, 2024-05-27:
```
12:15:30.464   ERROR utilsexc.cpp:873 Execution of command 'C:\Users\Mohamed El-Desoky\Desktop\portable opencpn5.9\opencpn-glutil.exe opengl-info C:\Users\Mohamed El-Desoky\Desktop\portable opencpn5.9\gl_caps.json' failed (error 0: The operation completed successfully.)
12:15:30.464 MESSAGE OCPNPlatform.cpp:729 OpenGL test utility failed with exit code -1
12:15:30.464 MESSAGE OCPNPlatform.cpp:911 OpenGL test complete.
12:15:30.464 MESSAGE OCPNPlatform.cpp:913 BuildGLCaps fails.
```

nohal, 2024-05-27:
Remove the space from "portable opencpn5.9" and I suppose you have a workaround.

nohal, 2024-05-27:
OH, you probab;y don't there's another space even in your username...

Mohamed-ElDesoky, 2024-05-27:
oh it works, sorry for inconvenience

nohal, 2024-05-27:
This is a real issue

Mohamed-ElDesoky, 2024-05-27:
yes, and i would like to add when i tried to clone the repos and get the project built using the documentation stated in the fast track, the scripts isn't executed and finally i moved all the folders to c:\opencpn and then the scripts get executed normally

commit 9e4c2122 (2024-05-27): Quote paths in opencpn-glutil invocation. Fixes #3938