OpenCPN/OpenCPN #3944: Bug:  Chart Panel > Bouy/Light Labels

rgleason, 2024-05-29:
**Describe the bug**
Bouy/Light Labels do not work.

**To Reproduce**

No Bouy/Light Labels
[image]

Yet when Show Text is turned on, Bouy/Light Labels show properly
[image]

A clear and concise description of what you expected to happen.
I expect  Show Text  to not toggle    Bouy/Light Labels!
I expect Show Bouy/Light Labels   to toggle    Bouy/Light Labels!

Windows 11  Opencpn 5.9 most recent.

bdbcat, 2024-05-30:
"Also I noticed that User Standard, ALL and STANDARD do operate somewhat differently with these two toggles."
How different?

rgleason, 2024-05-30:
> "Also I noticed that User Standard, ALL and STANDARD do operate somewhat differently with these two toggles."
> How different?  
Rick:  Not different at all.  User Standard, All and STANDARD  show the bouys, but not Basic.
The operation of Bouy/Light Labels is the same, and has the same problems.

After having a chanch to think about this further, in the interest of simplicty and flexibility I would prefer to see this solution:

- Completely Separate the operation of Show Text and Show Bouy/Light Labels, so that when one wants Bouy/Light labels (without all the Text), just turn on Bouy/Light labels.
- Additionally, one other change should be made. "Light descriptions" is subservient to "Lights" which is OK, but it should not be subservient to "Show Text". 

I think these two changes would really help users understand the Chart Panel Vector Chart settings much better and allow them to adjust the kinds of text they need quite easily.

rgleason, 2024-05-30:
After having gone through this process, the issue is changing the way OpenCPN has worked using the keyboard shortcut key "T" to turn on/off all Text.  The above approach requires the user to use the "T" for text, "K" for Light descriptions and "B" for Bouy/light Labels.

This might be too much, also it is too late in the game I think for all of these changes.

So we'd better just make it work right without leaving the user lost, or just not worry about having shortcuts right now for  Light Descriptions and Bouy/light labels, but try to do the best thing for Chart Panel which really is a quick and easy way adjust the chart for current needs.

rgleason, 2024-05-30:
So the Alternative is to keep as is, the keyboard shortcut "T" to toggle
- Text
- Light Descriptions
- Bouy/Light Labels

but have the dependencies greyed out when the will not work.
If   "Show Text" is  Unchecked,
Then "Show Light Descriptions"  (double dependency on "Lights")
and "Show Bouy/Light Labels should both be greyed out and unavailable.

rgleason, 2024-05-30:
I have no idea which is easier to do, but I favor the first approach, as it gives the user finer control over Text, and Text is not just lumped into one big thing.

bdbcat, 2024-05-31:
In implementation, "Show Text" is a master gate for rendering of all text, from whatever source.  It is a bit too late in this cycle to refactor that behavior.  Would affect NOAA S-57 charts, and require an immediate and concurrent update to o-charts plugin.  So not in the cards just now.

Will move to making "gray-out" states more consistent.

rgleason, 2024-05-31:
Ok , To compliment "T" for Text, for the future, : I've reserved the Keyboard shortcut keys for :
- K  Light Descriptions
- B  Bouy/Light Labels

Will label them as reserved (future use) in the Keyboard Shortcuts list and in Chart Panel > Vector Display 

Thanks.

bdbcat, 2024-06-01:
Corrected as discussed.  Also affects  #3945.
Please close both if satisfied.

rgleason, 2024-06-01:
Dave,  should I use opencpn_5.9.0-0+1579.5f6ba04_setup.exe 
because there appears to be a later version that does not have setup.exe for Windows?
https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/

I want to install the right one to test.

bdbcat, 2024-06-01:
opencpn_5.9.0-0+1579.5f6ba04_setup.exe is the latest CI build, last night, after my push.  So it should be right.
Where did you find a "later version that does not have setup.exe for Windows?"

rgleason, 2024-06-01:
Found them right here https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/
it looks like they are more recent and have a different number.

rgleason, 2024-06-01:
Dave, please look at your email, for an email from me.  re Marks vs Waypoints. Thanks

rgleason, 2024-06-01:
Dave, this works like a charm. Thank you.

closed 2024-06-01 by rgleason

bdbcat, 2024-06-01:
Rick.
"Dave, please look at your email, for an email from me. re Marks vs Waypoints. Thanks"
No related email received.

rgleason, 2024-06-02:
I've re-sent the email as a forward.