OpenCPN/OpenCPN #3945: Bug: Chart Panel > Light Descriptions

rgleason, 2024-05-29:
**Describe the bug**

"Show Light Descriptions" is subservient to both "Show Text"  and  "Show Lights" but the greyed out operation of "Show Light Descriptions"  is **not** greyed out when "Show Text" is turned OFF as it is greyed out when "Show Lights" is turned OFF.

Then Show Light Descriptions should not be UN greyed until both Show Text and Show Lights is are Turned ON.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: [e.g. macOS]
 - Version [e.g. 22]

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: [e.g. Galaxy S20]
 - Andorid version: [e.g. 10]

**Additional context**
Add any other context about the problem here. Screenshots are usually a good way to ilustrate problems, attach them if applicable.

rgleason, 2024-05-29:
After considering this problem and trying to use it. I realize now that  
"Show Light Descriptions"  should be controlled totally separately from "Show Text"

Show Light Descriptions should not be subservient to having Show Text turned ON.
Why?  There are important times we need Light Descriptions, but not have the confusion of other Text!

Therefore Light Descriptions should turn on and off independently.

bdbcat, 2024-05-30:
Well, maybe....

Anyone else have thoughts?

Hakansv, 2024-05-30:
Well, I can agree...
The light Descriptions would be dependent  on "[X] Lights" && "[X] Light Descriptions" and not dependent on "[X] Show text"

rgleason, 2024-05-30:
Yes, Please!!  Then I can finish that small part of the manual with a screenshot which is missing right now.
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:chart_panel:chart_panel_options:vector-charts#show_light_descriptions

rgleason, 2024-06-01:
Dave, works like a charm. Thank you!

closed 2024-06-01 by rgleason