OpenCPN/OpenCPN #3946: Bug: Anchoring Info works under ALL but not under USER STANDARD

rgleason, 2024-05-29:
**Describe the bug**

Anchoring Info works under ALL but not under USER STANDARD

The problem is probably due to ALL has turned on all layers, so the Show Anchoring Info toggle doew work.

However Anchoring Info under USER STANDARD does not have the mud and bottom layer turned on so that when its display is toggled, nothing shows.

**Expected behavior**
That it work.

bdbcat, 2024-05-30:
As designed.
In USER mode, if an object class is not enabled, it will not show.
IIUC, NotABug.

bdbcat, 2024-05-30:
More...
We should think of USER mode as "ENC Unleashed", or "Working without a net".
It is all too easy to set up an object display configuration that omits critical information, creating a dangerous situation.
I recommend a clear warning in the User Manual to suggest that USER mode is not meant for casual navigation, but reserved for special situations.
These categories (Base, Standard, All, User) are defined by IMO, with some careful thought as to the object classes displayed in each mode..  Most users should be using Standard mode for daily navigation.

rgleason, 2024-05-30:
I have augmented the manual with description of what will work to show Seabed Area object layer.  [Show Anchor Info](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:chart_panel:chart_panel_options:vector-charts#show_anchoring_info)
In User Standard mode, showing the seabed information requires that the Seabed Area Object layer be turned ON and that Show Text is turned ON.

In Chart Panel there is no "Show seabed area" button, so we can only make notes about it.

Also I have tried to make a Warning about User Standard, but perhaps there are some mixed signals in the writeup on this [Chart Panel > Display Categories page](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:chart_panel:chart_panel_options:display-category) that need to be improved.  I am sure it could be consolidated more, but I am getting burnt out.  Your thoughts?

closed 2024-05-30 by rgleason

bdbcat, 2024-05-30:
The link for display categories page looks very good as is.  No complaints.
I'm content.