OpenCPN/OpenCPN #3947: Bug: Chart Panel  Show Visible Sector Lights does not seem to be working.

rgleason, 2024-05-29:
**Describe the bug**
I cannot get Show Visible Sector Lights to work using ALL, Standard, or USER Standard.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

[image]

[image]

rgleason, 2024-05-29:
PS I'd like to show small screenshots for all of these Chart Panel  at each  Show ____

Hakansv, 2024-05-30:
"Show visible.." That's visible for the ship. The ship must be be within the range of a single sector to be lit without a "mouse over".
The light object must also have a noted sector range. I know not all have a range. Sometimes the explanation from the HO is: "there are so many obstructions, like islands, around so to set a range is not meaningful." 

Here the range is 9 Nm 
[image]

bdbcat, 2024-05-30:
Please close if satisfied.

rgleason, 2024-05-30:
This is not readily apparent to the user and needs to be flagged in the manual. I will close once I complete that.  The older manual had  illustrations without mentioning the distance.

rgleason, 2024-05-30:
What are Extended Light Sectors?  How are they activated now?
See  https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:set_options:charts:vector_charts#bouys-lights

Or is this old material that should be removed?

Hakansv, 2024-05-30:
I think "Extended Light Sectors" and leading sectors are rather well described as well as the method to make the leading sector(s) graphical, e.g. a "mouse over".
The "Show Visible sector lights" option is just another method to show these of possible interest nearby the ship.

rgleason, 2024-05-30:
1. Extended Light Sectors are not the same as "Visible Sector Lights", one shows as a beacon, the other is a segment of a circle.  The question here, is how are the Extended Light sectors  turned on?   What object layer?

Hakansv, 2024-05-30:
No, "Show visible..." shows all kind of sectors.
"Show lights" turn all lights on. Extended, leading, lights are indicated by dotted lines but are light up graphical  to it's nominal range by a "mouse over"

rgleason, 2024-05-30:
Hakan, using ALL, Standard and User Standard Modes I get the same thing, but it is not that: this is in Maine, Monroe Island, Owls Head Light.

Mouse Rollover with Lights and Light Description and Show visible sector lights.
The outside ring is shown on mouse rollover.
[image]

Mouse Rollover with Lights and Light Description on ring show on mouse rollover.
[image]

rgleason, 2024-05-30:
I think the Extended Sectors are supposed to show as a "Beacon" would show. Take a look at 

>  What are Extended Light Sectors? How are they activated now?
> See https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:set_options:charts:vector_charts#bouys-lights

bdbcat, 2024-05-30:
Now I am confused.
In standard nav-speak, a "beacon" is a possibly unlit fixed mark.  Think "channel marker".
So please restate why re-opened.

rgleason, 2024-05-31:
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:set_options:charts:vector_charts#bouys-lights
"Activating the Extended Light sectors clarifies the situation. All sectors are extended and leading white sectors are emphasized in yellow and extend the full nominal range of the light. Just hold the cursor over a light and the extended sectors are activated."

[image]

What is this light? I believe it is an Extended Light Sectior.   How is it turned on?

bdbcat, 2024-05-31:
I think I understand the point, now.
There is no longer any specific "Extended Light Sectors" option.  Display of "Extended Light sectors" on mouse rollover is now the default behavior.
So the text should be something like:
"Extended Light sectors clarifies the situation. All sectors are extended and leading white sectors are emphasized in yellow and extend the full nominal range of the light. Just hold the cursor over a light and the extended sectors are activated."

Thanks

rgleason, 2024-05-31:
Dave, I have made that change, but the trouble is, I have never found a lighthouse or light that does this, why?   Is it really working or not?

[image]

Hakansv, 2024-05-31:
I've no beacons with sectors. But two lighthouse examples if you like:

All lights and one leading sector mouse over not in range of the ship.
[image]

Only "Show visible sector lights"  The other lights are not visible.
[image]

bdbcat, 2024-05-31:
Portsmouth, NH.
[image]

rgleason, 2024-05-31:
Thank you,  Dave and Hakan.  It does work. I guess the Extended Light Sectors are far and few between for NOAA.  I will include some of these screenshots.

closed 2024-05-31 by rgleason