OpenCPN/OpenCPN #3949: Bug:  Chart Panel > Vector Charts > Show Visible Sector lights on Proximity (without mouseover)

rgleason, 2024-05-31:
**Describe the bug**
The title "Show Visible Sector Lights" misrepresents what actually is intended.
It should read
"Show Visible Sector Lights on Proximity (without Mouseover)."
or 
"Show Visible Sector Lights on Proximity"

See https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:chart_panel:chart_panel_options:vector-charts

bdbcat, 2024-05-31:
Not really.
"Show visible sector lights" means show sector lights that are theoretically "visible to navigator on vessel if she looks out the window".  That is, vessel is within sector, and within range.
Does not mean "visible on computer screen".

rgleason, 2024-06-01:
Dave that is not what meant or intended. I wast trying to shorten it meaningfully.
How about 
Show Visible Sector Lights (vessel in sector & range) ?
Show Visible Sector Lights (in sector & range) ?
Show Visible Sector Lights from vessel sector & range?

Is there enough room in the menu?
Maybe have a second line below it.
So that Users will have some clue about how it really works.

bdbcat, 2024-06-01:
Not practical.  Translations are always longer.  Terse is better, easier to translate, easier to fit.
How about elaborating on the exact function of this button in the User Manual?

rgleason, 2024-06-01:
I understand the problem. It is a complicated function.
Users see "Show Visible sector lights" and think that you just turn it on and something will show and I will learn what it does, 
Then they think perhaps if I roll my mouse over a light it will work,
but it does not work that way either.
Can we do this?
"Show Visible Sector Lights (Read)"

bdbcat, 2024-06-01:
Sorry, no.  That will make no sense in translation.
Be careful about assuming what users will think.  "might think" is probably a safer position.
At some point we must trust that users will consult the docs for things they do not organically understand.
IMHO, user might think: "Hmm, that button does nothing.  I wounder if the docs say anything about it?"
The doc link you provided is perfectly clear, to me.

closed 2024-06-01 by rgleason