OpenCPN/OpenCPN #3951: Keyboard shortcut Ctrl+, options

CaptainMorrison, 2024-06-01:
**Describe the bug**
When i press shortcut keyboard Options(Control+,   ) 
When i close the options windows, Menu with buttons dissapear.  And sometimes when i close the windows reappears window options.
When i restart openplotter it appears again

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: [e.g. macOS]
 - Version [e.g. 22]

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: [e.g. Galaxy S20]
 - Andorid version: [e.g. 10]

**Additional context**
Add any other context about the problem here. Screenshots are usually a good way to ilustrate problems, attach them if applicable.

Hakansv, 2024-06-04:
I think you've to be more specific. At least I don't really understand.
On my Rpi4 I've used the Ctrl , to open and close  Options and see no oddities.
"Menu with buttons dissapear"  Which menu do you refer to?  (Screenshot?)
"Restart Openplotter" ?? Is that a RPi reboot or do you mean stop and restart OpenCPN?

Version. Please describe Rpi version, "latest" is Bookworm,  and OpenCPN version, latest official is 5.8.4.

leamas, 2024-06-09:
@CaptainMorrison  Without more details it's not  possible to work with this issue and we will have to close it

leamas, 2024-06-11:
@CaptainMorrison : we cannot really handle this without more details. 

Closing.

Please feel free to add more info to the closed issue in which case it might be reopened

closed 2024-06-11 by leamas