OpenCPN/OpenCPN #3952: RFE Harmonize waypoints and marks

rgleason, 2024-06-02:
https://opencpn.zulipchat.com/#narrow/stream/437623-User-Manual---Editors/topic/Manage.20Routes.20.3E.20Waypoints.20-.20Tab.2C.20Properties.20.26.20Use

Suggestions from Manfred

https://opencpn.zulipchat.com/#narrow/stream/437623-User-Manual---Editors/topic/Manage.20Routes.20.3E.20Waypoints.20-.20Tab.2C.20Properties.20.26.20Use/near/441919003

This is an issue to accompany changes in UI that  bdbcat has made.

rgleason, 2024-06-02:
Dave, It seems to be working very nicely. Thank you!   I enclose some screenshots. The OwnShip waypoints/routes page might need an adjustment, but hopefully Manfred will respond.
Mark-triangle
[image]
Mark blue ball
[image]
Magenta-diamond in route as mark
[image]
003 in route as mark
[image]

Waypoint green ball
[image]
Route properties, right click to select waypoint  properties   (mark?)
[image]

rgleason, 2024-06-02:
Manfred's  TB > Optoins > Ship > Ownship > Routes-Points  notes
 
[image]
 
In Zulip, Manfed wrote:

The instances circled in green refer to points, which are part of a route.
The instances circled in orange refer to objects, which have been created by "Drop Mark"

To me, it makes a lot of sense to call the one waypoint, since in a route "I am on my way to this point", and the other mark, since it marks a point of interest in the chart.

I am aware that the two objects have different names in navobj, however here I am just asking as a user to have a nomenclature, where different things are called differently.

I believe that waypoints and marks are nice and clear terms, which should then be used consistently throughout the entire user interface.

As a matter of fact, there are only a few instances, which ask for changes (in the GUI only I hope):

rgleason, 2024-06-02:
Anything like this menu, dealing with Routes should use "waypoint" because the components are part of a  route.
[image]

ManfredRad, 2024-06-02:
Thanks very much for following here my suggestion. It looks pretty good to me and feels much cleaner.

There is one inconsistency: in Route Properties, when right clicking on a route point a menu appears with first (correct) entry: "Waypoint properties ...", however then a dialog box appears saying "Mark properties". 

This is definitely wrong, when the point is a route point only existing within a route. You could argue this is fine for a mark which became part of a route, though. I'd think it is good to be not too nit-picky here. 
My suggestion: No matter of nature of the route point, it is a point within a route and thus Waypoint Properties seems more appropriate.

[image]

[image]

ManfredRad, 2024-06-02:
As Rick pointed out: In the Options - Ships - Routes/Point dialog, there is an instance, where waypoint is used and mark should be used instead. (orange circles below)

And you could argue, that the lower area (circled in green), which applies to points in a route could be moved up so that all route related sections are together and the only mark related section is at the bottom of this dialog box.

[image]

ManfredRad, 2024-06-02:
One more minor point discovered: after dropping a new mark, the right click for undo/redo say waypoint, instead of undo create mark, redo create mark

[image]

[image]

bdbcat, 2024-06-03:
After latest patch, finished here, I think.

ManfredRad, 2024-06-03:
@bdbcat looks good to me. Thanks.

bdbcat, 2024-06-03:
Ready to close.

closed 2024-06-04 by rgleason

rgleason, 2024-06-05:
Thanks Dave.