OpenCPN/OpenCPN #3953: Change language doesn't update without restarting.

Hakansv, 2024-06-02:
**Describe the bug**
Earlier when changing language the updated was performed when leaving options window using OK.
Now I've to restart O to get the new language.
If this as desired we may reintroduce the old message to restart O?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Options->User.
2. Click on the Languages drop down and select English(US) or the system's language.
3. Click Apply->OK to close Options
4. The language is still the same as before until O is restarted.

**Expected behavior**
If intentional we may add a user message. Else the language would change when Options is closed

**Desktop (please complete the following information if applicable):**
 - OS: Windows 10
 - Version present Git master 5.9

Hakansv, 2024-06-03:
@bdbcat 
After above patch O crashes in Dashboard after select a another language -> OK.
When restarting after the crash the new language is though implemented also in Dashboard.

Call stack
```
>	dashboard_pi.dll!DashboardWindow::SendSentenceToAllInstruments(DASH_CAP st, double value, wxString unit) Line 5980	C++
 	dashboard_pi.dll!dashboard_pi::SendSentenceToAllInstruments(DASH_CAP st, double value, wxString unit) Line 966	C++
 	dashboard_pi.dll!dashboard_pi::HandleN2K_129029(ObservedEvt ev) Line 2323	C++
 	[Inline Frame] dashboard_pi.dll!dashboard_pi::Init::__l2::<lambda_427c4cbb2f90bf826fd31bdba6190b56>::operator()(ObservedEvt) Line 657	C++
 	dashboard_pi.dll!wxEventFunctorFunctor<wxEventTypeTag<ObservedEvt>,<lambda_427c4cbb2f90bf826fd31bdba6190b56>>::operator()(wxEvtHandler * __formal, wxEvent & event) Line 547	C++
 	[External Code]	
 	[Frames below may be incorrect and/or missing, no symbols loaded for wxbase32u_vc14x.dll]	
 	opencpn.exe!PluginLoader::UpdatePlugIns() Line 758	C++
 	opencpn.exe!ApplyLocale() Line 8274	C++
 	opencpn.exe!MyFrame::DoOptionsDialog() Line 4211	C++
 	opencpn.exe!MyFrame::DoSettings() Line 2839	C++
 	opencpn.exe!ChartCanvas::OnKeyDown(wxKeyEvent & event) Line 3083	C++
 	[External Code]	
 	opencpn.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nCmdShow) Line 737	C++
 	[External Code]	
```
[image]

bdbcat, 2024-06-03:
Please test with no NMEA or N2k or sK inputs active at the time of language change.

Hakansv, 2024-06-04:
Yes, verified. When having switching off all input streams the language change was performed wo obstacles an without the need to restart O to finalize the change.

Hakansv, 2024-06-04:
@bdbcat  Correction.
After having rebuilt O from scratch all is working as it should.
Now I can change language without problem also with active data streams.
Also Dashboard's language is updated on the fly.
Sorry for the trouble. Closing this issue.

bdbcat, 2024-06-04:
There is still an exposure, randomly hit.
From the stack trace we see:
1.  Language change causes a reload of plugins.
2. Meanwhile, comms are still active during the plugin reload.
3. If dashboard gets a comm message in a "bad" time, he could crash.
4. This does not happen at program startup, because comm messages are not sent to plugins until all is stabilized.

So, still a valid bug, rather hard to reproduce.

bdbcat, 2024-06-07:
Spent some time thinking how this bug could be induced.
I see no way in normal flow for a problem in initialization of Dashboard objects.  It looks like the dashboard instruments are all fully initialized (by ApplyConfig()) before the message lambdas are bound.  So there is no "bad time" for DB to get an N2k message.
So, OK to close this, pending further incidents in the field.

closed 2024-06-07 by bdbcat