OpenCPN/OpenCPN #3957: Scan Charts and Update Database is ignored on Options->OK

transmitterdan, 2024-06-03:
**Describe the bug**
Request to scan charts and update is ignored on Options->OK.

**To Reproduce**
Steps to reproduce the behavior:
1) Open Options dialog
2) Click "Scan Charts and Update Database"
3) Click OK
4) Dialog goes away but no chart folders are scanned.

If you click "Apply" button then the chart folders are scanned.

**Expected behavior**
Chart folders should be scanned either on OK or Apply.

 - OS: Windows
 - Version 11

transmitterdan, 2024-06-03:
I think this started on 21-May-2024.  There was some fiddling with how options dialog handles "OK" and now it skips some steps.

bdbcat, 2024-06-03:
Yes, just been looking at this regarding #3953.  Same trouble.  Several items skipped on OK click now.
Do you know the commit that started the problem (21 May) ?

closed 2024-06-04 by transmitterdan