OpenCPN/OpenCPN #3959: Preferences window losing focus

nohal, 2024-06-04:
After the latest changes to the behavior of the preferences window, it loses focus and disappears under the main app window on macOS

nohal, 2024-06-06:
@leamas https://github.com/leamas/OpenCPN/commit/8a2987c6b28c3095771dfd7235cb156e50941aa4 made the options window constantly sinking under the main window. Unfortunately even with `wxSTAY_ON_TOP`. I am running out of ideas how this could be fixed without reverting these patches. The niche use case of NMEA Debug window IMHO does not justify the horrible UX it causes for any interaction with options.
Yes, I know the NEMA Debug window behavior was weird, but this does not seem to be a good way to fix it.

leamas, 2024-06-06:
On phone... This seems to be about macOS only (?), windows and Linux are ok .

I have no current macOS vm running, hard to test.

Assume you have tested wxFRAME_FLOAT_ON_PARENT?

nohal, 2024-06-06:
It is not a `wxFrame` but a `wxDialog`, so `wxFRAME_FLOAT_ON_PARENT` is not usable

nohal, 2024-06-06:
And yes, it is about macOS only as all the other z-order issues

leamas, 2024-06-06:
On phone... This seems to be about macOS only (?), windows and Linux are ok .

I have no current macOS vm running, hard to test.

Assume you have tested wxFRAME_FLOAT_ON_PARENT?

Back later when i have a computer at hand

leamas, 2024-06-06:
...and you have tried setting focus using wxWindow::SettFocus()?

nohal, 2024-06-06:
SetFocus would not help, the main problem is the window does sink under the main frame again as soon as something happens.
On macOS we need to either be modal or be a `wxFrame` floating on parent.
But I found the reason for `wxSTAY_ON_TOP` being ignored. That makes the preferences usable even though not exactly the way we want.

closed 2024-06-06 by nohal

leamas, 2024-06-06:
I. e., the proper fix would be to use Frame instead of a dialog.  I can see that.

nohal, 2024-06-06:
The proper fix would probably be to completely decouple the NMEA debug window from the options dialog and make it modal again. But for now we can probably live with `wxSTAY_ON_TOP`

commit 6aa60cc9 (2024-06-06): Make options dialog stay on top of all windows on macOS

This somewhat mitigates the effect of the preferences window not being modal anymore.

For the next iteration more refactoring is required to decouple the NMEA debug window
from the options completely as `wxSTAY_ON_TOP` is suboptimal for maintaining the preferenes window visible.

Closes #3959