OpenCPN/OpenCPN #3961: Bug:  List Contents results in Route Tab only two route tab functions are able to do anything.

rgleason, 2024-06-04:
**Describe the bug**
Terry Slatter wrote:
I see that the List Contents checkbox causes the layer to appear in the **Route & Mark Manager > Route Tab.** But it seems that **only two Route Tab functions are able to do anything with the selected layer:** 
Center View and Activate/Deactivate. Interestingly, 
Reverse doesn't reverse the order in which Activate traverses the route. I think this is a deficiency.

See 
https://opencpn.zulipchat.com/#narrow/stream/437623-User-Manual---Editors

Hakansv, 2024-06-04:
A layer can't be edited in any way. That's by design.
If yo want to in any aspect be able to change marks contained in a gpx file you should not load the file as a layer but rather use the import function in Route & Mark manager.

nohal, 2024-06-04:
NOTABUG

closed 2024-06-04 by nohal