OpenCPN/OpenCPN #3962: Plugin manager produce a busy pointer all time Options is open.

Hakansv, 2024-06-05:
**Describe the bug**
A soon PIM is opened the busy pointer is shown all time as long Options is open

**To Reproduce**
Steps to reproduce the behavior:
1. Open Options. Click "Plugins" if not already active.
2. Busy pointer is shown.
3. Click another options tab. Busy is still active.
4. Click OK to close Options. Normal pointer is recovered.

**Expected behavior**
No busy pointer unless OCPN is really busy.

**Desktop (please complete the following information if applicable):**
 - OS: Win10
 - Version Today's Git v5.9.0

nohal, 2024-06-05:
Probably need to find what erroneously starts the spinner not calling `AbstractPlatform::ShowBusySpinner()` and fix that, I don`r see any problem in https://github.com/OpenCPN/OpenCPN/commit/0f81b3de2cd0f5c7711b03067d198723b2781d08.

Hakansv, 2024-06-05:
A very short test:
Here's a part of the call stack to 
void OCPNPlatform::ShowBusySpinner(void) {
  AbstractPlatform::ShowBusySpinner();
}
when I opened the PIM tab in Options.

```
>	opencpn.exe!OCPNPlatform::ShowBusySpinner() Line 1684	C++
 	opencpn.exe!LoadAllPlugIns(bool load_enabled) Line 691	C++
 	opencpn.exe!options::DoOnPageChange(unsigned int page) Line 8324	C++
 	opencpn.exe!options::OnPageChange(wxBookCtrlEvent & event) Line 8159	C++
 	wxbase32u_vc14x.dll!wxAppConsoleBase::HandleEvent(wxEvtHandler * handler, void(wxEvtHandler::*)(wxEvent &) func, wxEvent & event) Line 662	C++
 	wxbase32u_vc14x.dll!wxAppConsoleBase::CallEventHandler(wxEvtHandler * handler, wxEventFunctor & functor, wxEvent & event) Line 673	C++
 	wxbase32u_vc14x.dll!wxEvtHandler::ProcessEventIfMatchesId(const wxEventTableEntryBase & entry, wxEvtHandler * handler, wxEvent & event) Line 1484	C++
 	wxbase32u_vc14x.dll!wxEvtHandler::SearchDynamicEventTable(wxEvent & event) Line 1952	C++
 	wxbase32u_vc14x.dll!wxEvtHandler::TryHereOnly(wxEvent & event) Line 1675	C++
 	[Inline Frame] wxbase32u_vc14x.dll!wxEvtHandler::TryBeforeAndHere(wxEvent &) Line 4007	C++
 	[Inline Frame] wxbase32u_vc14x.dll!wxEvtHandler::ProcessEventLocally(wxEvent &) Line 1612	C++
 	wxbase32u_vc14x.dll!wxEvtHandler::ProcessEvent(wxEvent & event) Line 1585	C++
 	[External Code]	
 	[Frames below may be incorrect and/or missing, no symbols loaded for wxmsw32u_core_vc14x.dll]	
 	wxbase32u_vc14x.dll!wxAppConsoleBase::HandleEvent(wxEvtHandler * handler, void(wxEvtHandler::*)(wxEvent &) func, wxEvent & event) Line 662	C++
 	wxbase32u_vc14x.dll!wxAppConsoleBase::CallEventHandler(wxEvtHandler * handler, wxEventFunctor & functor, wxEvent & event) Line 673	C++
 	wxbase32u_vc14x.dll!wxEvtHandler::ProcessEventIfMatchesId(const wxEventTableEntryBase & entry, wxEvtHandler * handler, wxEvent & event) Line 1484	C++
 	wxbase32u_vc14x.dll!wxEventHashTable::HandleEvent(wxEvent & event, wxEvtHandler * self) Line 1087	C++
 	wxbase32u_vc14x.dll!wxEvtHandler::TryHereOnly(wxEvent & event) Line 1679	C++
 	[Inline Frame] wxbase32u_vc14x.dll!wxEvtHandler::TryBeforeAndHere(wxEvent &) Line 4007	C++
 	[Inline Frame] wxbase32u_vc14x.dll!wxEvtHandler::ProcessEventLocally(wxEvent &) Line 1612	C++
 	wxbase32u_vc14x.dll!wxEvtHandler::ProcessEvent(wxEvent & event) Line 1585	C++
 	[External Code]	
 	wxbase32u_vc14x.dll!wxEvtHandler::ProcessEvent(wxEvent & event) Line 1598	C++
 	wxbase32u_vc14x.dll!wxEvtHandler::SafelyProcessEvent(wxEvent & event) Line 1711	C++
 	[External Code]	
 	wxbase32u_vc14x.dll!wxAppConsoleBase::Dispatch() Line 409	C++
 	[External Code]	
 	wxbase32u_vc14x.dll!wxEventLoopBase::YieldFor(long eventsToProcess) Line 152	C++
 	wxbase32u_vc14x.dll!wxEventLoopBase::Yield(bool onlyIfNeeded) Line 112	C++
 	wxbase32u_vc14x.dll!wxAppConsoleBase::Yield(bool onlyIfNeeded) Line 420	C++
 	[External Code]	
 	opencpn.exe!MyFrame::DoOptionsDialog() Line 3978	C++
 	opencpn.exe!MyFrame::DoSettings() Line 2839	C++
 	opencpn.exe!ChartCanvas::OnKeyDown(wxKeyEvent & event) Line 3083	C++
 	wxbase32u_vc14x.dll!wxAppConsoleBase::HandleEvent(wxEvtHandler * handler, void(wxEvtHandler::*)(wxEvent &) func, wxEvent & event) Line 662	C++
 	wxbase32u_vc14x.dll!wxAppConsoleBase::CallEventHandler(wxEvtHandler * handler, wxEventFunctor & functor, wxEvent & event) Line 673	C++
```

Hakansv, 2024-06-05:
btw:

Shouldn't "Hidebusyspinner" hide it instead of show it?
See code
```
//--------------------------------------------------------------------------
//      Platform Display Support
//--------------------------------------------------------------------------

void OCPNPlatform::ShowBusySpinner(void) {
  AbstractPlatform::ShowBusySpinner();
}

void OCPNPlatform::HideBusySpinner(void) {
  AbstractPlatform::ShowBusySpinner();
}
```

nohal, 2024-06-05:
Yes, of course :facepalm:

Hakansv, 2024-06-05:
Yes, tested "HideBusySpinner" instead and then it works. 
You fix? :)

nohal, 2024-06-05:
I did

closed 2024-06-05 by Hakansv