OpenCPN/OpenCPN #3965: Chart hamburger not working in OpenCPN  5.9. on Mac

ManfredRad, 2024-06-05:
When clicking to the chart hamburger icon (lower left) no chart panel option dialog appears. Zoom in/out, chart scale and follow are working.

OpenCPN 5.8.4 is working correctly. This bug description applies only to 5.9. and is persistent since a couple of weeks. 

Here is a screen from 5.8.4 for illustrative purposes:
[image]

**Desktop (please complete the following information if applicable):**
 MacOS Sonoma 14.5 on MacBook Air
OpenCPN 5.9.0-universal+9ad1e2db3

Checked disabling OpenGL (did not help). Tried also the other package (5.9.0-legacyintel+9ad1e2db3) for Mac: no succes

bdbcat, 2024-06-05:
I suppose you mean "lower right" ?

ManfredRad, 2024-06-05:
Ah, yes, the other starboard. Lower right it is.

rgleason, 2024-06-05:
https://opencpn.zulipchat.com/#narrow/stream/432618-OCPN-5.2E10-Release-plan/topic/MacOS.20Chart.20Panel.20Hamburger.20.28bug.3F.29

bdbcat, 2024-06-06:
Close?

ManfredRad, 2024-06-06:
Has been reconfirmed by another user. So, seems to be a bug.

Hakansv, 2024-06-06:
Manfred..
Did you tried the new build?. As seen above a correction was made three hours ago.

ManfredRad, 2024-06-06:
Hakan:

OpenCPN_5.9.0-legacyintel+9514.64f20b75a-12.pkg works

OpenCPN_5.9.0-universal+9502.dc21fe5f9-14.pkg does not work

Both files on cloudsmith are 3hours old, but I guess it is different builds.

Will check the next universal build as well as soon as it is available.

I guess it is fixed.

Manfred

rgleason, 2024-06-06:
Manfred,  If this bug is confirmed to be fixed by someone here then the OP original person should close it. Thanks. ( Keep in mind that the builds are different, but I think the universal build should work on all MacOS including the M.)

nohal, 2024-06-06:
Confirmed fixed.

closed 2024-06-06 by nohal