OpenCPN/OpenCPN #3966: Isolated Danger Mark in ENCs not visible (any more)

ManfredRad, 2024-06-06:
Isolated dangers with a depth below the safety depth in an area of safe depth should be marked by magenta cross (see chart no 1, K11 and other instances).

This seemed to be built in previous versions of OpenCPN, but is not working anymore in 5.9 or 5.8.4 on Mac or Pi.

Here a figure from the old manual (currently made not visible, since the feature is not available).

[image]

Below is a current screen shot with wrecks at a depth  of 2.9m, 3.5m and 1.4m. The first two are located  in an area with more than 10m depth. 
Settings used: deep water 10m, safe depth 5m, shallow depth 2m. Display category: all; all check boxes turned on, except "show text" since it will print "WK" right on top of the depth value (see second screen shot)

[image]

Same area with text visible

[image]

rgleason, 2024-06-06:
Manfred please give lat/long or location, is is Noaa?

ManfredRad, 2024-06-06:
Positions added in original post.

bdbcat, 2024-06-06:
And what chart set, please?

ManfredRad, 2024-06-06:
I tried with whatever I can get a hold of. The above example in Chesapeake was created using old cm charts.

Here is the same position using current NOAA ENCs:

[image]

The other example was done with current o-charts Germany.

bdbcat, 2024-06-07:
Corrected, please close if satisfied.

ManfredRad, 2024-06-07:
Looks good to me. Thanks. I'll close the issue.

closed 2024-06-07 by ManfredRad