OpenCPN/OpenCPN #3969: Dashboard PI Speedometer shows STW label, but has SOG values

torrmundi, 2024-06-06:
**Describe the bug**
The Speedometer shows a label "STW".  The values shown by the speedometer correspond to the SOG values.
[image]

**To Reproduce**
Create a dashboard with Speedometer and SOG and STW. 
Note which (SOG or STW) corresponds to the speedometer value.

**Expected behavior**
Either the label should be SOG, or the value should be STW.

**Screenshots**
See attached screenshot.

**Desktop (please complete the following information if applicable):**
 - OS: Win11 home

Hakansv, 2024-06-06:
As of your shot the speedometer shows SOG in the graph and as extra info STW by its value.
The "STW" designation for the value would be clear in that instrument?

If you want also a "Speed though water meter" graphical instrument or another design of the present Speedometer I'd call it a future request and not a bug.

torrmundi, 2024-06-07:
Hakan, of course you are correct. I spent a day sailing with the speedometer open and never noticed that STW is an additional value. I thought it was a label.  So, the nature of my issue is different: Speedometer is ambiguous. SOG and STW are both "speedometers". Perhaps a better label would be STW?

Hakansv, 2024-06-07:
The Speedometer is a graphical dial showing SOG. The extra STW value is a text only instrument.
I can agree the position and size of the STW value are not nice and even disturbing? Could we discuss improvements instead?

torrmundi, 2024-06-07:
Sorry, tired after arriving in USA from Bahamas. 
Perhaps it would be better to relabel the widget not as a speedometer, which is quite obvious, but to relabel it as "Speed - SOG" or simply "SOG"? 
Here's a possible improvment, but I'm not sure if it will scale well or not:
[image]

Hakansv, 2024-06-07:
I personally think your sketch is nice and clear.
How to develop it in Dashboard I'm not sure about. Someone?
If such a change is feasible for next O update is also not clear just now.

torrmundi, 2024-06-08:
It looks great! That addresses everything I brought up.

Hakansv, 2024-06-08:
@bdbcat 
Thoughts? Would you accept a late PR for 5.10?

bdbcat, 2024-06-08:
I'll take it.  Looks very nice.

PR #3975 (2024-06-08): D-board Adjust extra STW appearance on SOG-Speedometer
Closes #3969
closed 2024-06-08 by bdbcat