OpenCPN/OpenCPN #3972: Combine routes deletes secondary route

tcs1026, 2024-06-07:
**Describe the bug**
The documentation on Combine Routes says that the secondary route remains unchanged. This is when moving the last waypoint of the primary route onto the first waypoint of the secondary route.
Observed in MacOS 5.9.0-universal+fee4e5477.

**To Reproduce**
Steps to reproduce the behavior:
1. Display two routes.
2. move the last waypoint of Route A onto the first waypoint of Route B. Click Yes to both popups.
3. Open the Routes & Marks Manager. Find that Route B is no longer listed.

**Expected behavior**
The manual (https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:route_manager:use_routes) and discussion (https://github.com/OpenCPN/OpenCPN/pull/2375#issuecomment-922271803) says that Route B is unchanged.

When I added screenshots to the manual, I stated that Route B is removed. This may need to change if this is a bug that is fixed.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
MacOS 14.5 Sonoma on M2 chipset

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: [e.g. Galaxy S20]
 - Andorid version: [e.g. 10]

**Additional context**
Add any other context about the problem here. Screenshots are usually a good way to ilustrate problems, attach them if applicable.

rgleason, 2024-06-07:
Terry, very good write up.  I am sorry to have you do this. I only had my phone, now I am at computer and have run a number of tests combining and merging routes.  Without really knowing the full intention here, I think removing the other end of the route is the right thing to do, but I don't know how it picks which end, do you?   Is there a way to undo it I wonder.

rgleason, 2024-06-07:
LATER: Is it deleting the whole route for you on a Mac???
It isn't doing that for me on Windows. One part of the route is kept and the other part is removed when the two routes are merged.
I am trying to understand what determines which part is kept and which part is removed.

rgleason, 2024-06-07:
See subsequent discussion and discovery process.
https://opencpn.zulipchat.com/#narrow/stream/437623-User-Manual---Editors/topic/Use.20Routes.20-.203.20Example.20Screenshots.20needed

bdbcat, 2024-06-08:
In the specific case you describe:
Indeed, Route B is assimilated as part of Route A, and so no longer exists uniquely.
As designed.

rgleason, 2024-06-08:
@tcs1026  Dave has responded to the question. Now the OP (original poster) (That's you Terry) need to Close this Issue.
Also, since that is how the function was designed, we need to change the text in the manual. Can you do that please? Thanks. Rick

tcs1026, 2024-06-08:
I'll make the changes to the doc later today and close this issue then.

tcs1026, 2024-06-08:
Edited the content in the manual's Use Routes page to explain the process of combining routes and adding a detour.

closed 2024-06-08 by tcs1026