OpenCPN/OpenCPN #3973: Vector Charts - 4color and safety depth contour

ManfredRad, 2024-06-07:
There are two issues with the placement of the safety depth contour and the use of the four colours  in vector charts.

Bug 1: safety contour (contour line, which is printed bold) is always at 5fm
Bug 2: colouring is off by one contour line when using integer values

Chart no1 suggests to use two colours below the safety contour and two above. The safety contour shall be printed thicker than the other contours. See screenshot from Chart No1

[image]

I added the text in red, denoting shallow water, safe water and deep water. And there is one area between shallow and safe, which I called in between.

Using  1fm, 3fm, 6fm I get the following result in the current NOAA chart of the Chesapeake Bay. This chart uses Fathoms as its native unit. So we see contour lines at 1fm, 2fm, 3fm, 5fm, 6fm , 10fm, ....labelled in red in the screen shot below, and four colours used: dark blue 0..2, light blue 2..5, safe 5..10, wight 10...infinity. Basically: all colours are off by one contour line

[image]
The safety contour line sits at the 5fm line, the shallow water area extends to 2fm, the deep water area does not start at 6fm, but at 10fm (not visible)

I can play tricks and type in the values 0.9fm, 2.9fm, 5.9fm instead to get the desired colouring:
[image]

Notabene: the transition in the text color of the depth soundings (black below safety contour, grey above safety contour) is in both cases right. The transition occurs at the 3fm contour line.

The above screen shots are generated at the following position:

[image]

bdbcat, 2024-06-08:
Regarding the safety contour:
So, this turns out to be a rounding error, introduced by the cartographer.
All S57 chart depth information is encoded in the cell in units of meters.  It is the international standard.
For the chart shown, the "3 fm" contour has been encoded in the cell as "5.400000" meters, by choice of the cartographer.  Note that this value is a little bit less than 3 fm.
The safety depth set by the user in OCPN is converted to meters with full precision.  In this case, 3 fm is calculated to be 5.4864 meters.
The test to identify the safety contour is carried out in native S57 coordinates, or meters.
Result:  The safety depth is greater than the selected contour, so the next deeper contour is used, in the case 5 fm.
In this particular case, the difference is about 86 mm, or about three inches.
Now, we might say 3 inches in 18 feet is well within our navigational tolerance, considering tides, bottom composition, etc.  And we might be right.  So we might be tempted to say, for the case of safety contours only, that +/- 2% is "close enough".
But if we start making approximations here, we are very soon lead to the question: "Now what did the user actually intend, and how can we be sure?"

Comments invited.

ManfredRad, 2024-06-08:
I see the point. I was assuming that OpenCPN internally calculates using the unit of the chart. But, on second thought I see that it is much easier going to metric. And it is the right way to go, since all, but 3, countries in the entire world, use the metric system. It is no real issue - for most of us. 

And, as you said, for all practical navigational uses, it is "close enough", when considering wave height, dependency of sea level on wind and accuracy of depth soundings.

Thanks for the clarification and I know now, what to write in the manual. 

Since this behaviour of OpenCPN is in fact not a bug, but a well known feature when converting imperial to metric, the issue can be closed here.

closed 2024-06-08 by ManfredRad