OpenCPN/OpenCPN #3977: Crash when opening Preferences

apezio, 2024-06-08:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: [e.g. macOS]
 - Version [e.g. 22]

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: [e.g. Galaxy S20]
 - Andorid version: [e.g. 10]

**Additional context**
Add any other context about the problem here. Screenshots are usually a good way to ilustrate problems, attach them if applicable.

bcn2, 2024-06-10:
What happens if you work with an empty/new config file?

bdbcat, 2024-06-10:
From the crash report, we see that OCPN is failing in an area that has received substantial updates in the current 5.9.1 Beta test series.
Recommend that you install O5.9.1 Beta now, and retest.
https://www.cruisersforum.com/forums/f134/opencpn-v5-9-1-beta-test-starts-286845.html

Thanks
Dave

apezio, 2024-06-10:
In 5.8.4, the first time I delete  ⁨~/Library⁩/Preferences⁩/opencpn⁩/opencpn.ini I can open the Preferences, but after a quit and start it crashes again until I delete the prefs again (and to reiterate - that only works once).

I will trythe 5.9.1 beta...

apezio, 2024-06-10:
Since there is an active beta, I suppose this page could use an update:  https://opencpn.org/OpenCPN/info/beta.html

apezio, 2024-06-10:
Looks like the beta requires at least Mac OS 11 (but like many, I stick with 10.14 for it's native 32-bit support).  So no luck there.

nohal, 2024-06-10:
> Looks like the beta requires at least Mac OS 11 (but like many, I stick with 10.14 for it's native 32-bit support). So no luck there.

https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/detail/raw/OpenCPN_5.9.1-legacyintel%252B9630.b67b62772-12.pkg/-/

apezio, 2024-06-10:
5.9.1 hangs upon clicking "Next" at the first (or only?) 'Initial Configuration' step/window.  After a force quit, it opens and crashes immediately.

I restored my original 'Preferences/opencpn' folder from 5.6 / 5.2 and no more crash in 5.9.1.

I tried deleting the prefs fodler again, and once more 5.9.1 hangs at the initial config window.  So I am guessing that the hang, and subsequent force quiting is corrupting the prefs, making 5.9.1 crash ever yime it opens thereafter.

So a new user might see 5.9.1 hang and then crash constantly thereafter, while a use with an older prefs folder may not see an issue.

Hope this info helps and I'm happy to try and narrow it down more if anyone thinks it would be useful or has some suggestions on how to do so.   For now, 5.9.1 seems stable using my old prefs folder (and I can open the Preferences!!) :)

nohal, 2024-06-10:
Does it really hang during the initial config wizard or were you just impatient? Try waiting at least a minute, it seems that on macOS the events don't propagate well during the network connection scan so it seems like it is stuck, but eventually continues for me.
I don't have a macOS system old enough to test...

apezio, 2024-06-10:
You are correct, after a few minutes (I did not time it) it does get to the next page of the setup window.  However, upon completion, opening the prefernces still induces a crash.  I left everything at the default values and did not add any chart folders.

Further attempts to open 5.9.1 cause an immediate crash.

I restored my original prefs folder (which was working a few minutes ago) and now 5.9.1 crashes immediately upon opening 100% of the time.

nohal, 2024-06-10:
Can you attach the zipped contents of `~/Library/Preferences/opencpn` that cause the crash? I can't reproduce it neither killing the app during the wizard nor finishing it.

nohal, 2024-06-10:
If macOS generates the crash report for you, it would also be interesting to see.

apezio, 2024-06-10:
The Initial config window hangs for approx. 300 seconds.  During this time Activity Monitor reports OpenCPN using <1% CPU.  It could be related to 'gpsd', which the next page shows as something OpenCPN identifies as a GPS source?  But - after the 300 second hang,  the second page hangs for about 15 seconds, after which 'GPSd' shows up as a source.  So on second thought that I think the 15 second hang is searching for GPS/USB sources and the 300 second is uknown.

Prefs folder and two crash reports for 5.9.1 attached.  

From crash reports:

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [1134]

[opencpn.zip](https://github.com/user-attachments/files/15779919/opencpn.zip)
[opencpn5.9.1-crash-reports.zip](https://github.com/user-attachments/files/15779932/opencpn5.9.1-crash-reports.zip)

If I try to open 5.9.1 with default prefs (or any prefs now), 4/5 times it crashes immediately.  The one time it doesn't crash, opening prefs works. But after quiting, the next time I open 5.9.1 it always crashes immediately.

It's truly frustrating how inconsistent the behavior is.  I try the exact sequence of opening and closing 5.9.1 a dozen times, and I can't find a pattern.

apezio, 2024-06-10:
After all the crashing, Opening 5.2.0 reports it is deleting a stale lock file, which none of the other versons report doing.  I wonder if the lock file could have something to do with all of this.

apezio, 2024-06-10:
This is the prefs folder after sucessfully opening 5.2.0, 5.6.4 and 5.8.4 (in that order). 5.9.1 still crashes immediately.

[opencpn-after5.2.0-5.6.4-5.8.x.zip](https://github.com/user-attachments/files/15779976/opencpn-after5.2.0-5.6.4-5.8.x.zip)

FYI even 9.2.1 Safe Mode crashes 4/5 times.  The only thing that is consistent is 5.2.0 not crashing :)

nohal, 2024-06-10:
Hmm, none of that is reproducible on my machine, all working fine. But it is an M1 running under Rosetta, the Intel one died... I will eventually try to get a 10.14 VM running somehow, but so far no joy.

bdbcat, 2025-04-26:
OBE