OpenCPN/OpenCPN #3982: Toolbar Plugin icons

TwoCanPlugIn, 2024-06-10:
Given the parameters of the Plugin API

extern "C" DECL_EXP int InsertPlugInToolSVG(
    wxString label, wxString SVGfile, wxString SVGfileRollover,
    wxString SVGfileToggled, wxItemKind kind, wxString shortHelp,
    wxString longHelp, wxObject *clientData, int position, int tool_sel,
    opencpn_plugin *pplugin);

In the same way that a plugin's toolbar icon varies depending on whether the plugin is inactive or active (toggled or normal), I would have assumed that the SVGfileRollover image would be displayed whenever the user "hovers" over the toolbar icon. I don't think I have witnessed what I assume to be the intended behaviour over several iterations of plugins and different versions of OpenCPN.

Also what is the purpose of the parameters, label, & longHelp ? shortHelp is the string displayed when the user hovers over the toolbar icon, but what is the purpose of both label & longHelp ?

I don't care whether this is fixed or not, nor am I aware of others reporting the problem.

Is this another YAPI ? (Yet another partial feature implementation)

nohal, 2024-06-10:
It is yet another "The API did not change for backward compatibility, although some parameters are ignored in later versions of OpenCPN".

closed 2024-06-10 by nohal

TwoCanPlugIn, 2024-06-10:
Or put another way "The API was broken and we hoped no one would notice"

Just like this [one](https://github.com/OpenCPN/OpenCPN/issues/3941)

A bit cavalier.

nohal, 2024-06-10:
No, nothing to do with that one.