OpenCPN/OpenCPN #3987: Chart Downloader: bogus feedback

leamas, 2024-06-12:
**Describe the bug**
Running the downloader it consistently reports downloading 0MB/0MB despita that is actually is downloading data

**To Reproduce**
Steps to reproduce the behavior:
1. Open   Options | Charts, select the Downloader tab
2. Click "Add Catalog"
3. Find for example the Alaska charts _AK - Alaska_
4. Accept the proposed download directory
5. Update the new directory
6. Click _Download selected charts_ button
7 See  0MB/0MB feedback
8 After the download is complete the  new chart directory is available and can be added to database.

**Expected behavior**
Feedback should be relevant, at least the total size.
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - Debian Bookworm + Orange Pi 5 kernel patches

**Smartphone/Tablet (please complete the following information if applicable):**
 - Orange Pi 5 [enhanced RPi 5](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-5.html)

leamas, 2024-06-12:
@mgrouch  @saillog ^

mgrouch, 2024-06-13:
For me it simply hangs OpenCPN. I’ve posted report on forum. I can’t even kill OpenCPN regularly. I had to use kill -9

bdbcat, 2024-06-13:
mgouch...
I don't think it hangs.  Try selecting and download two or three charts.  While the dialog does not update, the actual download completes.  ergo, some trouble with dialog signaling.
If you select all charts, and wait long enough, it will complete, I think.

bdbcat, 2024-06-13:
Fixed, tested on linux.

Corsair-63, 2024-06-13:
RPI4 (X11) & O 5.9.1 (neither openplotter nor Flatpack)
when done the same procedure I can see the progress of the download and sometimes the system seems to be hanged after some minutes it respond again but nobody knows if charts installed (well you know if you search in the specific folder)

leamas, 2024-06-13:
It's spelled Flatpak ;)

Corsair-63, 2024-06-13:
forced to do one chart download, it seems to be freeze for someeeee seconds (supposed the downloading time) after that it respond again, but no progress display.
 
this is the screenshot while downloading, and no response while supposed downloading file.
[image]

leamas, 2024-06-13:
Rebuilt master at 4c59033ee locally on the Orange Pi.  The feedback is mostly the same 0MB/0MB although I can see occasional glimpses of a non-zero total.

Might be  a performance issue. My SBC has a faster chip than even RPi5 and not least an SSD disk.  OTOH, I assume that when tested on "Linux" this was on a desktop machine which certainly has better throughput.

leamas, 2024-06-13:
Did a quick hack increasing the 20ms sleep introduced in 4c59033ee  to 50ms. Basically no change. It seems like it provides better (not good) data at start, but it becomes worse over time when downloading more files..

Looking at the code it looks polled rather than event-driven even though the download is done in a separate thread. That is, the main thread continues in the dialog  although it starts a separate download thread. Might be wrong, just a quick look.

bdbcat, 2024-06-14:
If someone has the time...
Please test this function on O584, on rPI, with fast internet speed.
Thanks
Dave

rgleason, 2024-06-14:
Dave, the only builds that I see are from 17 hours ago.  
If you want me to test with O584, I can also do that.
Ok, I've tested with O584 by starting download of Inland Charts.
Using O584  

1. I am easily and nicely able to scroll up and down to see the charts being downloaded while they are being downloaded.
2. I am able to mouse click outside the Chart Download menu and nothing happens, no change of state, grey out, or freeze, or anything.
3. It is well behaved and offers no potential questions or confusion by users about what happened..

[image]

https://github.com/OpenCPN/OpenCPN/issues/3990#issuecomment-2167101084

leamas, 2024-06-18:
I have reverted the changes aimed to fix #3923 which caused this bug. This basically means that we have the old, broken interface it the NMEA window but also that this issue should be fixed. Will do some testing later, time permitting.

rgleason, 2024-06-18:
Maybe this should be deffered until later if that is any help?

leamas, 2024-06-18:
Well, it's hopefully fixed.  The downside is that that #3923 is reopened.

closed 2024-06-30 by nohal