OpenCPN/OpenCPN #3990: ChartDownloader  When Updating charts, any mouse action causes greyed screen and pause issues.

rgleason, 2024-06-13:
**Describe the bug**
This one is a little hard to explain.
1. From Chartdownloader, Update a chart catalog and start to Download it.
2. Move the mouse away from the Chartdownloader screen and left click.
3. The Chartdownloader menu goes grey and there is a big pause.
4. Wait.
5. The screen finally changes and all the downloads have been completed.
6. This is a little strange to users. Why not have a message,  Cancel download? Yes/No
Also
1. From Chartdownloader, Update a chart catalog and start to Download it.
2. Move the mouse away from the Chartdownloader screen and try to scroll down the list of charts.
3. The Chartdownloader menu goes grey and there is a big pause.
4. Wait.
5. The screen finally changes and all the downloads have been completed.
6. This is a little strange to users. Why not have a message,  Cancel download? Yes/No

**To Reproduce**
As described.

**Expected behavior**
6. This is a little strange to users. Why not have a message,  Cancel download? Yes/No

Win11  Most current from CS not released

leamas, 2024-06-13:
Well, if you click or scroll outside the window it loses focus, but keeps downloading. What is the problem here besides the missing download feedback (#3987)?

If user wants to cancel the download there is a Cancel button.

rgleason, 2024-06-13:
Yes, it looses focus and keeps downloading.  This is a strange state for users and they will be confused when it happens. The screen gets kind of frozen and they will start hitting other keys rather than just waiting.

PS It may be working slightly differently in your Linux setup.  This is a windows thing I suspect..

leamas, 2024-06-13:
Not a bug.

Corsair-63, 2024-06-13:
it is the same behavior with Raspi bookworm, not only related to windows, and in the course of operation no display of downloaded and remaining size.

leamas, 2024-06-13:
>  and in the course of operation no display of downloaded and remaining size.

Yes, this is the real issue here, filed as #3987. 

Besides that, I see no bug -- that windows loses focus when we click outside of it is normal. So is a download happening in background in a window without focus.

rgleason, 2024-06-13:
There should be a message, as I stated earlier. Users will be confused.

leamas, 2024-06-14:
Not a nug, perhaps an RFE

TwoCanPlugIn, 2024-06-20:
I'm unsure whether this is still open, but in its current form is unsatisfactory.

Ultimately the download in the background succeeds, but when OpenCPN and/or the chart downloder loses focus, this is not the UI to present to the user.
[image]

transmitterdan, 2024-06-20:
What OS is this? On Windows 11 when downloader is downloading and loses focus it does not do that.  It happily keeps reporting the download progress as it did when in focus.

leamas, 2024-06-20:
There are changes in master handling this error. Please test a build from git or wait for next beta. Real issue is #3987

Corsair-63, 2024-06-20:
I use not to have activated the chart plugin network because I use o-charts, but I've done another test.

Windows 11 23H2 ARM64 - Surface X  SQ1 16 GB RAM.
O 5.9.1
activated chart downloaded plugin, added chart catalogue OSMSHP and downloaded all charts:
first step, since to be all working fine:
[image]
after some seconds:
[image]
and general display:
[image]

I've been this time patient enough and looking at places if everything is working.
for the time being I'm more than ten minutes and still O hasn't recovered the functionality.

but I've seen that OneDrive has activity and seen that, I've checked that charts are being downloaded to the setup folder because I see busy the OneDrive over such folder.

at this time around 15 minutes and seen still downloading high resolution chart and no recovered the functionality of O.

Corsair-63, 2024-06-20:
in above procedure left the computer until the end of transfer, it took about 45 minutes and at last given a transfer error in the full resolution map, but the rest were downloaded, and O makes responsive and showing the correct data.
after that I downloaded the big one and I left the computer running and downloading, husband/father tasks, when I come back the files was downloaded and installed.
I left the computer when 55 Mb was downloaded and still not lost the plot the screen.

rgleason, 2024-06-23:
Sorry, cant test waiting out bad weather today.

leamas, 2025-07-18:
This should be fixed in 5.12. Unless there is more input here this issue will be closed.

closed 2026-02-11 by leamas