OpenCPN/OpenCPN #3992: OCPN beta crashes with Measure tool

antipole2, 2024-06-14:
**Describe the bug**
On MacOS with 5.9.1-universal+b67b62772, using the Measure tool crashes OCPN.

**To Reproduce**
Steps to reproduce the behaviour:
1. Option click on the canvas and select Measure
2. Click the pencil tool on the canvas to position where to measure from
3. Crash

For me this is reproducible.

**Crash dump**
[OCPN crash Measure tool.txt.zip](https://github.com/user-attachments/files/15842034/OCPN.crash.Measure.tool.txt.zip)

**Desktop (please complete the following information if applicable):**
 - OS: [e.g. macOS] Mac M2
 - Version 14.4.1

antipole2, 2024-06-15:
This no longer happens with latest build fc6f5dd.
Will close once official beta also clear.

antipole2, 2024-06-18:
Ignore my comment this no longer happening.
I am getting this crash every time I try to use the Measure tool.

bdbcat, 2024-06-18:
@bcn2...
Reproducible for you?

bcn2, 2024-06-19:
No, Measure works without issues on the M1 with 14.5
Tested with UI in English, Spanish and German.
(With OCPN-Draw enabled, it crashes when changing the language  - a new, different story)

bdbcat, 2024-06-20:
Tony...
Please retest, with all plugins disabled.

antipole2, 2024-06-20:
> Tony... Please retest, with all plugins disabled.

Same with all plugins disabled

antipole2, 2024-07-02:
No longer occurs in Beta 2 - great!

closed 2024-07-02 by antipole2