OpenCPN/OpenCPN #3995: Dashboard PI does not respect user choice of units for Depth

torrmundi, 2024-06-15:
**Describe the bug**
The user selects depth units under Options->Display.>Units>Depth.
When the Depth or Depth gauge are added to a Dashboard, the units used are always meters.

**Expected behavior**
The units of depth in the Dashboard PI should be according the the user's setting in Options.

**Desktop (please complete the following information if applicable):**
 - OS: windows 11
 - Version: opencpn_5.9.0-0+1618.c283089_setup.exe

Hakansv, 2024-06-16:
It's not a bug. The function is as designed. In Dashboard all units are set in Dashboard's Preferences->Appearance. 
Please change the Dashboard settings to suit your preferences and close this if it works? 

One can possibly imagine a future Dashboard development to add a depth unit as the speed unit "Honor OCPN settings" ?

TwoCanPlugIn, 2024-06-16:
The Dasboard has its own separate settings.

However one little bug, In the Apparent Wind Angle and Speed, the true wind does not respect the units ?
[image]

Hakansv, 2024-06-16:
@TwoCanPlugIn 
Strange! Compare my pictures.

Win10:
[image]
RPi5:
[image]

TwoCanPlugIn, 2024-06-16:
Ooops.

You're right (as usual).

I forgot to restart O or the Dashboard during my quick & dirty test.