OpenCPN/OpenCPN #4000: When OpenCPN window is maximized at shutdown, it gets resized to 0x0 during next start

nohal, 2024-06-17:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Maximize OpenCPN
2. Close it
3. Start it again
4. The window is shown as small as the WM allows instead of maximized

**Expected behavior**
Window maximized on next start

**Desktop (please complete the following information if applicable):**
 - Gnome 45

**Additional context**

I have added some debug output to illustrate what happens during the startup

```
MyFrame::ODoSetSize x: 1344 y: 697
MyFrame::ODoSetSize x: 1344 y: 665
MyFrame::ODoSetSize x: 1344 y: 665
MyFrame::ODoSetSize x: 1344 y: 665
MyFrame::ODoSetSize x: 310 y: 1
*** BUG ***
In pixman_region32_init_rect: Invalid rectangle passed
Set a breakpoint on '_pixman_log_error' to debug

*** BUG ***
In pixman_region32_init_rect: Invalid rectangle passed
Set a breakpoint on '_pixman_log_error' to debug

MyFrame::ODoSetSize x: 310 y: 1
*** BUG ***
In pixman_region32_init_rect: Invalid rectangle passed
Set a breakpoint on '_pixman_log_error' to debug

MyFrame::ODoSetSize x: 310 y: 1
```

bdbcat, 2024-06-18:
Fixed for me, tested on vm.

nohal, 2024-06-18:
Yes, looks OK

closed 2024-06-18 by nohal