OpenCPN/OpenCPN #4005: Reconsider unarr support

bgermann, 2024-06-19:
**Describe the bug**
libarchive has improved RAR support significantly since unarr was introduced.
#963 claims that unarr was introduced because of it having RARv5 support, which it does not have actually.
Libarchive has added RARv5. The main thing that is still missing compared with unarr is RARVM filters (which is relevant for comic book format cbr mainly).

Maybe it is time to drop unarr.

leamas, 2024-06-19:
I cannot see that this is a bug. Relabeling.

nohal, 2024-06-19:
Did you confirm libarchive can now successfuly decompress the Peruvian charts from https://www.dhn.mil.pe/portal/cartas-raster?

bgermann, 2024-06-19:
I have just tested them all. The only file that libarchive has an issue with is 5371 Portulano Pucallpa y Cercanías:
SBS_5371/PORT_PUCALLPA01.KAP: File CRC error: Invalid or incomplete multibyte or wide character

Comparing the output with 6 bytes differing from the expected uncompressed file.

nohal, 2024-06-19:
Thanks, means we can drop `unarr` and use `libarchive` for all the archive formats as those were the only reason we couldn't.

bgermann, 2024-06-19:
That was with v3.7.2. The 6 bytes are fixed with the latest 3.7.4, which has https://github.com/libarchive/libarchive/pull/2124 included. So yes, unarr can be dropped.

bgermann, 2024-06-22:
Generally, 3.6.0 is the first version that [supports RAR filters](https://github.com/libarchive/libarchive/pull/1503) which are needed for the Peruvian charts.

leamas, 2024-08-20:
>  So yes, unarr can be dropped.
 The 6 bytes are fixed with the latest 3.7.4,

Which basically means that it can be removed when 3.7.4 is packaged. I still does not see it any any OS I use, but guess it will land in Fedora 41, and later on in sid/trixie.

We should prepare for 3.7.4 landing in sid. I'm on it.

bgermann, 2024-08-20:
Debian trixie and sid have 3.7.4 already. That is why the can be [fixed in Debian](https://bugs.debian.org/1078767) now.

leamas, 2024-08-20:
> Debian trixie and sid have 3.7.4 already.

Indeed, I stand corrected.

nohal, 2024-08-21:
This issue still needs some additional love on macOS and Windows...

leamas, 2024-08-21:
That as a little terse, so to speak. Could you  please expand?

nohal, 2024-08-21:
The dependency bundles need updating. I will do it as usually, just until this done, this issue is not to be closed.

leamas, 2024-08-21:
Ah, ok. But it is really no hurry in this context. The build does the right thing whether libarchive is 3.7.4+ or not, falling back to in most cases bundled unarr leaving us with no external deps.

leamas, 2024-08-21:
@bgermann : I will drop the unarr dep in next opencpn upload.

leamas, 2025-04-29:
This is now handled by the cmake setup, basically using libarchive if recent enough else falling back to unarr. 

Specifically, the unarr dep has been removed from the ci packaging which eventually will turn up in sid and trixie(-backports)

Closing

closed 2025-04-29 by leamas