OpenCPN/OpenCPN #4008: Girb UI Project position course and speed values missing. Speed field size need Win zoom factor.

Hakansv, 2024-06-21:
**Describe the bug**
Values for Project position are missing when grib UI is opened.
At Windows scaled size = 175% the speed field is to small so it's hard to read the value.
**To Reproduce**
Steps to reproduce the behavior:
1. Be sure to navigate so own COG and SOG are present.
2. Click on Grib tool
3. Project position values would have been present but course is "0" and speed is default 6.0 knots.
4. In this example the course should have been about 45° and speed 9.7 knots
5. Also is the speed field not showing more than the first digit and the unit is hidden.

**Expected behavior**
Project position should have actual COG and SOG values. Now one have to enter both values manual.
Adaptation to Windows display scaling.

**Screenshots**
Two shots. First at Windows scale 175% and a second at scale 100%

**Desktop (please complete the following information if applicable):**
 - OS: Windows 10 @ recommended 175% scale.
 - Version Present Git master 5.9.1

[image]
[image]

rgleason, 2024-06-22:
Reccommended scale for my lapttop is higher due to higher dpi. Will this scale properly?

Hakansv, 2024-06-22:
Additional info of the course and speed values.
It turns out I've sometimes opened the Grib UI before a valid (slowly updated)SOG was received and then no data.
If I wait long enough before opening Grib the data is there so this part is not a bug but an ask for enhancement to update the values more times. Maybe when the "Project position" is enabled?
The display bug is though still valid.
We could make this an after 5.10 issue.

bdbcat, 2024-06-24:
Corrected layout, supports MSW scaling.

closed 2024-06-24 by Hakansv