OpenCPN/OpenCPN #4018: Memory leaks: valgrind detects blocks are definitely lost

sebastien-rosset, 2024-06-28:
**Describe the bug**
valgrind detects blocks are definitely lost.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile and install OpenCPN, head of master
2. Run OpenCPN. enable plugins (list provided below). Exit.
3. Run OpenCPN with valgrind: `valgrind --leak-check=full opencpn | tee opencpn.log`. Wait until the UI is fully displayed.
4. Click Navigate -> "Exit OpenCPN"
5. Check valgrind output, in particular messages that state "blocks are definitely lost" 

**Expected behavior**
Memory blocks should not be lost.

**Screenshots**
See attached valgrind log

**Desktop (please complete the following information if applicable):**
 - OS: Ubuntu 24.04
 - Version: OpenCPN 5.9 compiled from head of master.

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: N/A
 - Android version: N/A

**Additional context**

Below is an example of blocks that are definitely lost. The full log is attached.
```
==6270== 72,168 (72,056 direct, 112 indirect) bytes in 1 blocks are definitely lost in loss record 20,984 of 21,026
==6270==    at 0x4846FA3: operator new(unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==6270==    by 0x3C75E6AC: ???
==6270==    by 0x3C715502: ???
==6270==    by 0xA4B953: PluginLoader::LoadPluginCandidate(wxString const&, bool) (plugin_loader.cpp:560)
==6270==    by 0xA4D7CE: PluginLoader::LoadPlugInDirectory(wxString const&, bool) (plugin_loader.cpp:687)
==6270==    by 0xA4EB4F: PluginLoader::LoadAllPlugIns(bool, bool) (plugin_loader.cpp:426)
==6270==    by 0x696D6B: MyFrame::OnInitTimer(wxTimerEvent&) (ocpn_frame.cpp:4777)
==6270==    by 0x5729011: wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (in /usr/lib/x86_64-linux-gnu/libwx_baseu-3.2.so.0.2.2)
==6270==    by 0x5729792: wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) (in /usr/lib/x86_64-linux-gnu/libwx_baseu-3.2.so.0.2.2)
==6270==    by 0x572B4F5: wxEvtHandler::TryHereOnly(wxEvent&) (in /usr/lib/x86_64-linux-gnu/libwx_baseu-3.2.so.0.2.2)
==6270==    by 0x572B57E: wxEvtHandler::ProcessEventLocally(wxEvent&) (in /usr/lib/x86_64-linux-gnu/libwx_baseu-3.2.so.0.2.2)
==6270==    by 0x572B691: wxEvtHandler::ProcessEvent(wxEvent&) (in /usr/lib/x86_64-linux-gnu/libwx_baseu-3.2.so.0.2.2)
==6270==
```

nohal, 2024-06-30:
I somewhat tend to think these are actually false positives caused by Valgrind not understanding the fact that the plugins are going to be destroyed by calling their `destroy_pi()` via `pic->m_destroy_fn`, but did not check very thoroughly.

bdbcat, 2024-07-03:
Deferred to OCPN 5.12.0

leamas, 2026-02-11:
Situation in the 5.14 beta1 is similar: Blocks are lost in several places including LoadPlugin, pango and GL.

However, to state that this is a bug requires more in depth analysis. The real problem is perhaps that we don't use use valgrind as we should.

Closing this for now.