OpenCPN/OpenCPN #4019: Create Route Bug

Carl-Port, 2024-06-28:
When selecting the route line type, for example dot-dash is simply shown as dash.
Having created some route, when zooming in, the route disappears partially or completely.

Using Windows 10 pro

OpenCPN 5.9.1

bdbcat, 2024-06-29:
Pictures, please?

transmitterdan, 2024-06-29:
@bdbcat I can reproduce this on Win 11.  The "dot dash" selection just produces dash.
[image]

If I zoom in:
[image]
If I keep zooming in the dashed line goes away completely.  If the line type is solid, then it does not disappear.

Both of these are pretty minor buglets IMO.  I can look into it some.

bdbcat, 2024-06-30:
Corrected the style line rendering, two issues:
1.  "dot dash" format removed, not supported by current GL shader model.  Possible RFE for later version.
2. Corrected GL line rendering at very small scale.

bdbcat, 2024-06-30:
Please close this issue as resolved with respect to styled line rendering.  Too many other unrelated issues and suggestions for one task.
Thanks

rgleason, 2024-06-30:
Original poster, please close. Otherwise...

closed 2024-06-30 by nohal