OpenCPN/OpenCPN #4021: Corrupted File Explorer - Export or Import in Route & Mark Manager

Carl-Port, 2024-06-29:
Corrupted File Explorer - Export or Import in Route & Mark Manager

On Windows 10 Pro

OpenCPN 5.9.1

_____________________________

Opening the Route & Mark Manager, after selecting some GPX file, or even selecting several files, and trying to export or import any file, the folder referring to the Windows File Explorer does not correctly show the locations to be selected where to export or import, this problem already occurred in the previous version of OpenCPN.

Occasionally folders are shown, but usually only the folder names appear, and the selection commands in the bars appear totally corrupted.

The image below illustrates the problem.
_____________________________

[image]

Hakansv, 2024-06-29:
Car-Port..
If I understand you correct I don't see a problem. You have to select where on the file system you want save to file.
Basically OCPN ask Windows for the last folder you used to copy a file. If you agree save it. If not change to another folder. Most probably next time you want to export a file the first suggestion from Windows is the folder you selected last.

Carl-Port, 2024-06-29:
Dear Mr. Hakansv 

Well, I'm perfectly familiar with the mechanics of OpenCPN's file saving, importing and exporting system, but if you couldn't see the problem despite the image I provided, then I suggest you open your Windows' File Explorer and see its options and appearance in the bottom bar.

If I change my Windows' dark theme to the light theme, the problem apparently goes away, as far as I've been able to verify. 

So the incomplete way the export and import window is shown, maybe it's linked to the dark and light theme of Windows, but that's just a note of mine.

Thanks

Hakansv, 2024-06-29:
Well, I see no difference in any changed color scheme.
In your picture there are no left side folders list if that's your issue?
I can't say I've seen that on my W10 but I've have not changed Win's color scheme. And I don't think OCPN can tell Windows to not show the folders list. I would be a Windows setting?

bdbcat, 2024-06-30:
NOTABUG.
Some OS trouble with light/dark theme.  Not controlled by OpenCPN.

Carl-Port, 2024-06-30:
The fact that someone can't reproduce the problem on another computer doesn't mean that the problem doesn't exist, but I'm not going to dwell on it.

closed 2024-06-30 by Carl-Port

nohal, 2024-06-30:
The situation is a little different, nobody is able to reproduce the problem on any other computer... FWIW the problem you report does not materialize on any of my Windows 10 systems, all of which are using dark theme.