OpenCPN/OpenCPN #4022: ConnectionsDialog::OnSize called before the dialog is actually initialised

nohal, 2024-06-30:
Address sanitizer report:

```
AddressSanitizer:DEADLYSIGNAL
=================================================================
==1601==ERROR: AddressSanitizer: SEGV on unknown address 0xffffd847d7d9d7d7 (pc 0x0001012ad194 bp 0x00016f34f7e0 sp 0x00016f34f7e0 T0)
==1601==The signal is caused by a READ memory access.
    #0 0x1012ad194 in ConnectionsDialog::OnSize(wxSizeEvent&) connections_dialog.cpp:89
    #1 0x103b79028 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)+0x6c (libwx_baseu-3.2.0.2.3.dylib:arm64+0xdd028)
    #2 0x103b7a36c in wxEvtHandler::SearchDynamicEventTable(wxEvent&)+0x7c (libwx_baseu-3.2.0.2.3.dylib:arm64+0xde36c)
    #3 0x103b7a284 in wxEvtHandler::TryHereOnly(wxEvent&)+0x2c (libwx_baseu-3.2.0.2.3.dylib:arm64+0xde284)
    #4 0x103b7a1a0 in wxEvtHandler::ProcessEventLocally(wxEvent&)+0x2c (libwx_baseu-3.2.0.2.3.dylib:arm64+0xde1a0)
    #5 0x103b7a054 in wxEvtHandler::ProcessEvent(wxEvent&)+0x34 (libwx_baseu-3.2.0.2.3.dylib:arm64+0xde054)
    #6 0x1040f4158 in wxScrollHelperEvtHandler::ProcessEvent(wxEvent&)+0x2c (libwx_osx_cocoau_core-3.2.0.2.3.dylib:arm64+0x1dc158)
    #7 0x103b79e84 in wxEvtHandler::SafelyProcessEvent(wxEvent&)+0x14 (libwx_baseu-3.2.0.2.3.dylib:arm64+0xdde84)
    #8 0x1040b4c9c in wxWindowBase::SendSizeEvent(int)+0x7c (libwx_osx_cocoau_core-3.2.0.2.3.dylib:arm64+0x19cc9c)
    #9 0x1040f59b0 in wxScrollHelper::AdjustScrollbars()+0x16c (libwx_osx_cocoau_core-3.2.0.2.3.dylib:arm64+0x1dd9b0)
    #10 0x1040f55a0 in wxScrollHelperBase::ScrollDoSetVirtualSize(int, int)+0x24 (libwx_osx_cocoau_core-3.2.0.2.3.dylib:arm64+0x1dd5a0)
    #11 0x1012a29b0 in ConnectionsDialog::Init() connections_dialog.cpp:110
    #12 0x1012ac37c in ConnectionsDialog::ConnectionsDialog(wxScrolledWindow*, options*) connections_dialog.cpp:62
    #13 0x1015b0058 in options::CreatePanel_NMEA(unsigned long, int, int) options.cpp:1861
    #14 0x1015a95e4 in options::CreateControls() options.cpp:5847
    #15 0x1015a5a78 in options::options(wxWindow*, int, wxString const&, wxPoint const&, wxSize const&, long) options.cpp:1505
    #16 0x1014ac944 in MyFrame::OnInitTimer(wxTimerEvent&) ocpn_frame.cpp:4903
    #17 0x103b79028 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)+0x6c (libwx_baseu-3.2.0.2.3.dylib:arm64+0xdd028)
    #18 0x103b78ef0 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)+0x88 (libwx_baseu-3.2.0.2.3.dylib:arm64+0xdcef0)
    #19 0x103b7a2a4 in wxEvtHandler::TryHereOnly(wxEvent&)+0x4c (libwx_baseu-3.2.0.2.3.dylib:arm64+0xde2a4)
    #20 0x103b7a1a0 in wxEvtHandler::ProcessEventLocally(wxEvent&)+0x2c (libwx_baseu-3.2.0.2.3.dylib:arm64+0xde1a0)
    #21 0x103b7a054 in wxEvtHandler::ProcessEvent(wxEvent&)+0x34 (libwx_baseu-3.2.0.2.3.dylib:arm64+0xde054)
    #22 0x103b79e84 in wxEvtHandler::SafelyProcessEvent(wxEvent&)+0x14 (libwx_baseu-3.2.0.2.3.dylib:arm64+0xdde84)
    #23 0x103b29d78 in wxTimerImpl::SendEvent()+0x28 (libwx_baseu-3.2.0.2.3.dylib:arm64+0x8dd78)
    #24 0x18e37a5b4 in __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__+0x1c (CoreFoundation:arm64e+0x9a5b4)
    #25 0x18e37a258 in __CFRunLoopDoTimer+0x3c8 (CoreFoundation:arm64e+0x9a258)
    #26 0x18e379d90 in __CFRunLoopDoTimers+0x160 (CoreFoundation:arm64e+0x99d90)
    #27 0x18e35d1c8 in __CFRunLoopRun+0x73c (CoreFoundation:arm64e+0x7d1c8)
    #28 0x18e35c430 in CFRunLoopRunSpecific+0x25c (CoreFoundation:arm64e+0x7c430)
    #29 0x18f490a84 in -[NSRunLoop(NSRunLoop) runMode:beforeDate:]+0xd0 (Foundation:arm64e+0x5ba84)
    #30 0x103fbf3dc in wxGUIEventLoop::DoDispatchTimeout(unsigned long)+0x8c (libwx_osx_cocoau_core-3.2.0.2.3.dylib:arm64+0xa73dc)
    #31 0x103b59fe4 in wxCFEventLoop::DispatchTimeout(unsigned long)+0x28 (libwx_baseu-3.2.0.2.3.dylib:arm64+0xbdfe4)
    #32 0x103b5a124 in wxCFEventLoop::OSXDoRun()+0x6c (libwx_baseu-3.2.0.2.3.dylib:arm64+0xbe124)
    #33 0x103fbf8ac in wxModalEventLoop::OSXDoRun()+0x54 (libwx_osx_cocoau_core-3.2.0.2.3.dylib:arm64+0xa78ac)
    #34 0x103b5a18c in wxCFEventLoop::DoRun()+0x24 (libwx_baseu-3.2.0.2.3.dylib:arm64+0xbe18c)
    #35 0x103ad1a78 in wxEventLoopBase::Run()+0xbc (libwx_baseu-3.2.0.2.3.dylib:arm64+0x35a78)
    #36 0x103f331f0 in wxDialog::ShowModal()+0x60 (libwx_osx_cocoau_core-3.2.0.2.3.dylib:arm64+0x1b1f0)
    #37 0x101348c30 in wxEventFunctorFunctor<wxEventTypeTag<ObservedEvt>, LoadErrorsDlgCtrl::LoadErrorsDlgCtrl(wxWindow*)::$_0>::operator()(wxEvtHandler*, wxEvent&) event.h:547
    #38 0x103b79028 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)+0x6c (libwx_baseu-3.2.0.2.3.dylib:arm64+0xdd028)
    #39 0x103b7a36c in wxEvtHandler::SearchDynamicEventTable(wxEvent&)+0x7c (libwx_baseu-3.2.0.2.3.dylib:arm64+0xde36c)
    #40 0x103b7a284 in wxEvtHandler::TryHereOnly(wxEvent&)+0x2c (libwx_baseu-3.2.0.2.3.dylib:arm64+0xde284)
    #41 0x103b7a1a0 in wxEvtHandler::ProcessEventLocally(wxEvent&)+0x2c (libwx_baseu-3.2.0.2.3.dylib:arm64+0xde1a0)
    #42 0x103b7a054 in wxEvtHandler::ProcessEvent(wxEvent&)+0x34 (libwx_baseu-3.2.0.2.3.dylib:arm64+0xde054)
    #43 0x103b79e84 in wxEvtHandler::SafelyProcessEvent(wxEvent&)+0x14 (libwx_baseu-3.2.0.2.3.dylib:arm64+0xdde84)
    #44 0x103b79c58 in wxEvtHandler::ProcessPendingEvents()+0xf4 (libwx_baseu-3.2.0.2.3.dylib:arm64+0xddc58)
    #45 0x103aa742c in wxAppConsoleBase::ProcessPendingEvents()+0xdc (libwx_baseu-3.2.0.2.3.dylib:arm64+0xb42c)
    #46 0x103b59b94 in wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int)+0x38 (libwx_baseu-3.2.0.2.3.dylib:arm64+0xbdb94)
    #47 0x18e35d878 in __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__+0x20 (CoreFoundation:arm64e+0x7d878)
    #48 0x18e35d764 in __CFRunLoopDoObservers+0x214 (CoreFoundation:arm64e+0x7d764)
    #49 0x18e35cd90 in __CFRunLoopRun+0x304 (CoreFoundation:arm64e+0x7cd90)
    #50 0x18e35c430 in CFRunLoopRunSpecific+0x25c (CoreFoundation:arm64e+0x7c430)
    #51 0x198b00198 in RunCurrentEventLoopInMode+0x120 (HIToolbox:arm64e+0x33198)
    #52 0x198affe28 in ReceiveNextEventCommon+0xd8 (HIToolbox:arm64e+0x32e28)
    #53 0x198affd2c in _BlockUntilNextEventMatchingListInModeWithFilter+0x48 (HIToolbox:arm64e+0x32d2c)
    #54 0x191bbbd64 in _DPSNextEvent+0x290 (AppKit:arm64e+0x39d64)
    #55 0x1923b1804 in -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:]+0x2b8 (AppKit:arm64e+0x82f804)
    #56 0x191baf098 in -[NSApplication run]+0x1d8 (AppKit:arm64e+0x2d098)
    #57 0x103fbf50c in wxGUIEventLoop::OSXDoRun()+0x6c (libwx_osx_cocoau_core-3.2.0.2.3.dylib:arm64+0xa750c)
    #58 0x103b5a18c in wxCFEventLoop::DoRun()+0x24 (libwx_baseu-3.2.0.2.3.dylib:arm64+0xbe18c)
    #59 0x103ad1a78 in wxEventLoopBase::Run()+0xbc (libwx_baseu-3.2.0.2.3.dylib:arm64+0x35a78)
    #60 0x103aa6ab8 in wxAppConsoleBase::MainLoop()+0x5c (libwx_baseu-3.2.0.2.3.dylib:arm64+0xaab8)
    #61 0x103afb778 in wxEntry(int&, wchar_t**)+0x60 (libwx_baseu-3.2.0.2.3.dylib:arm64+0x5f778)
    #62 0x1014511f0 in main ocpn_app.cpp:745
    #63 0x18def60dc  (<unknown module>)

==1601==Register values:
 x[0] = 0xbebebebebebebebe   x[1] = 0x000000016f34f938   x[2] = 0x00000001012ad164   x[3] = 0x0000000000000000
 x[4] = 0x000000016f34f938   x[5] = 0x0000000103aa7648   x[6] = 0x000000016f34eb94   x[7] = 0x000000016f34eb88
 x[8] = 0x17d7d7d7d7d7d7d7   x[9] = 0x0000007000020000  x[10] = 0x00000000226853d5  x[11] = 0x00000000000001ff
x[12] = 0x000000000000002c  x[13] = 0x000000011dc0a3b0  x[14] = 0x3a0000010434cbf5  x[15] = 0x000000010434cbf0
x[16] = 0x0000000103b79f90  x[17] = 0x050e000191c1c7d4  x[18] = 0x0000000000000000  x[19] = 0x000000016f34f938
x[20] = 0x000000016f34f938  x[21] = 0x000000013c5c5c80  x[22] = 0x0000000000000003  x[23] = 0x0000000113429ef0
x[24] = 0x0000000000000000  x[25] = 0x000000013c5c5fdc  x[26] = 0x0000000000000014  x[27] = 0x0000001400000014
x[28] = 0x0000000000000000     fp = 0x000000016f34f7e0     lr = 0x0000000103b7902c     sp = 0x000000016f34f7e0
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV connections_dialog.cpp:89 in ConnectionsDialog::OnSize(wxSizeEvent&)
==1601==ABORTING
[1]    1601 abort      ./OpenCPN.app/Contents/MacOS/OpenCPN
```

bdbcat, 2024-06-30:
Is this a hard fault, or intermittent on MacOS?

nohal, 2024-06-30:
It is a hard fault with sanitizers enabled as, intentionally, nothing is automagically initialised to `NULL`. Everywhere else it depends on the compiler.

If you don't want to move the `Bind`, this will also do:

```diff
diff --git a/gui/src/connections_dialog.cpp b/gui/src/connections_dialog.cpp
index 9889882b0..794c45a94 100644
--- a/gui/src/connections_dialog.cpp
+++ b/gui/src/connections_dialog.cpp
@@ -92,6 +92,7 @@ void ::ConnectionsDialog::OnSize(wxSizeEvent &ev){
 void ConnectionsDialog::Init() {
   // Setup some initial values
   mSelectedConnection = NULL;
+  m_sbSizerLB = nullptr;

   // Create the UI
```

bdbcat, 2024-06-30:
OK, I understand.
No problem to delay the Bind().

PR #4023 (2024-06-30): Postpone binding ConnectionsDialog::OnSize event after the dialog is initialized
Closes #4022
closed 2024-06-30 by bdbcat