OpenCPN/OpenCPN #4027: beta2: FTBFS on Ubuntu armhf builders

leamas, 2024-07-01:
**Describe the bug**

FTBFS on Ubuntu armhf builders: 
```
/usr/src/opencpn-1:5.9.2-beta2+dfsg-1ubuntu1~bpo26.04.2/libs/mipmap/src/mipmap.c:169:(.text+0x3fc): undefined reference to `MipMap_24_neon'
/usr/bin/ld: /usr/src/opencpn-1:5.9.2-beta2+dfsg-1ubuntu1~bpo26.04.2/libs/mipmap/src/mipmap.c:169:(.text+0x404): undefined reference to `MipMap_32_neon'
```
**To Reproduce**
-  Create a source package according to https://opencpn-manuals.github.io/main/opencpn-dev/create-deb-package.html
- Push this package to a PPA

**Expected behavior**
Builds on all platforms

**Additional context**
Build logs: https://launchpadlibrarian.net/737585216/buildlog_ubuntu-noble-armhf.opencpn_1%3A5.9.2-beta2+dfsg-1ubuntu1~bpo26.04.2_BUILDING.txt.gz

Overall build status: https://launchpad.net/~leamas-alec/+archive/ubuntu/opencpn/+packages

leamas, 2024-07-02:
From the build log
```
-- Looking for __ARM_NEON__
-- Looking for __ARM_NEON__ - not found
```

... which probably means that while these symbols are referenced they are not built.

leamas, 2024-07-06:
In *libs/mipmap/src/mipmap_neon.c*

```
#if defined(__ARM_NEON) || defined(__ARM_NEON_FP)
#include <arm_neon.h>
```
This encloses the whole file.  One assumption is that neither __ARM_NEON nor __ARM_NEON_FP is defined on the failing buils, otherwise the missing symbols should be defined.

It seems that these are defined depending on various compiler options. We don't define any of these, and the compilation target capabilities is thus up to the compiler. It might certainly mean some  random defaults.  The question then  boils down to what kind of arm32 targets  we are actually compiling for.

See: https://gcc.gnu.org/onlinedocs/gcc/ARM-Options.html

leamas, 2024-07-06:
Indeed; this patch builds:
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index f88cb01f8..ba75be0d4 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -452,6 +452,11 @@ if (QT_ANDROID)
   set(CMAKE_SHARED_LINKER_FLAGS "-Wl,-soname,libgorp.so -Wl,--build-id")
 endif ()
 
+if (ARCH STREQUAL armhf)
+  # See: https://forums.raspberrypi.com/viewtopic.php?t=144115
+  add_compile_options(-mcpu=cortex-a53  -mfpu=neon-fp-armv8)
+endif ()   # Assuming at least RPi 3
+
 message(STATUS "Final compiler options:")
 message(STATUS "CMAKE_CXX_FLAGS: ${CMAKE_CXX_FLAGS}")
 message(STATUS "CMAKE_CXX_FLAGS_DEBUG: ${CMAKE_CXX_FLAGS_DEBUG}")

```

This patch assumes that we build for armhf binaries for hw similar to RPi 3 or better. Is this a reasonable assumption?

@bdbcat : thoughts?

nohal, 2024-07-06:
Probably yes, everybody using anything that is not a RPi with OpenCPN, if there actually are such people, should be prepared to suffer...

bdbcat, 2024-07-06:
"Is this a reasonable assumption?"
Yes.  Frankly, RPi3 is ancient history for our target users.  Running modern OCPN on RPi3 is suffering, indeed.

leamas, 2024-07-06:
>  Running modern OCPN on 32 bit RPi3 is suffering, indeed.

Agreed, and we should IMHO sooner or later drop the armhf builds, preferably sooner. However, will merge this patch for now.

commit 4f1c60cd (2024-07-06): armhf: Add target compiler options (#4027).

Add target options corresponding to a Raspberry Pi 3 or armhf builds.
This makes some things like foating  point hw and neon graphics
available, but will also fail on hw not on pair woth RPi 3.

Closes: #4027
closed 2024-07-06 by leamas